ut-issl/s2e-core #7: Move constant variables

200km, 2022-01-04:
## Overview
Move constant variables to the global environment.

## Details
Currently, constant physical variables are defined in several codes.  
They should be organized and defined in one code file.  
`GlobalEnvironment` class is suitable for the role.

## Conditions for close
When all constant variables are moved to a suitable place.

## Supplement
NA.

## Note
- Write a conclusion when closing the issue.

200km, 2022-03-31:
The math constants like 'pi' were moved to 'Library/math/Constant.hpp' by [this PR](https://github.com/ut-issl/s2e-core/pull/52).

200km, 2022-03-31:
@sksat 光速や太陽距離、地球半径などの物理定数系も数学定数と同じように実装する方針でよいでしょうか。

場所をどこにするのかは難しいですが、次のどちらかかなと思っています。    
- Library/physics/PhysicalConstants.hpp
- Environment/Global/PhysicalConstants.hpp

普通は前者かもですが、S2Eでは天体位置情報など似たものがGlobalEnvironmentとして定義されているので、この場所に置くのもそれなりに親和性があると思っています。いかがでしょう？

sksat, 2022-03-31:
そうですね．場所はどちらでもよいと思います(後者でも全然よさそうです)．

200km, 2022-03-31:
では後者で実装を進めようと思います。

200km, 2022-03-31:
MATH_CONSTANTS?
- [ ] deg2rad
- [ ] arcsec2rad
- [ ] rpm2rad_s
- [ ] rad_s2rpm

PHYSICAL_CONSTANTS?
- [ ] Speed of light in vacuume
- [ ] Boltzmann constants

Get from CSPICE?
- [ ] Earth Radius (average)
- [ ] Earth Gravity Constants

200km, 2022-04-07:
We merged #76, and close this issue.

closed 2022-04-07 by 200km