ut-issl/s2e-core #9: Modify place and structure of initializing functions

200km, 2022-01-04:
## Overview
Modify place and structure of initializing functions.

## Details
Currently, all codes of initializing functions are in the `InitInput` directory, but it is not appropriate.  

## Conditions for close
When all initializing codes are moved to the suitable place.

## Supplement
NA.

## Note
- Write a conclusion when closing the issue.

200km, 2022-04-15:
これもそろそろ手を付けようかなと思っています。  
それなりに小さな改修にしようと思うと、次のような流れかなと思いますが。どう思いますか？
- とりあえず[このディレクトリ](https://github.com/ut-issl/s2e-core/tree/develop/src/Interface/InitInput)は削除して、それぞれの初期化関数をそれぞれのクラス定義ファイルの近くに異動させる。

sksat, 2022-04-15:
これって，ファイル(名)→各種クラス，をやる関数群がディレクトリ切られてたくさん置いてある，ということですよね．コンストラクタにファイル名を突っ込むのは無意味にクラスのコードが肥大化してしまうのでかなり避けたいです．
一方で，ファイルを読んでクラスを作るのはコンストラクタ，というかnew的な操作であることは確かで，各々の構造を直接弄っているので管理のしやすさから単純に近いディレクトリ・ファイルにある存在しているべきだとは思います(かつ，抽象化されているとはいえファイルを読むという操作はその構造(クラス)になんら関わりが無いので，完全に同じ場所にはいてほしくない)．
なので，1つ目が一番いいと思います．そして地上局のコンストラクタがファイル読んでるならそれは分割したいですね．

200km, 2022-04-16:
コメントありがとうございます！  

> ファイル(名)→各種クラス，をやる関数群がディレクトリ切られてたくさん置いてある

そうですね。この部分が分かりづらいとか、汚いとこれまでに色んな人に不満を言われて来ているので修正したいなという感じです。

> コンストラクタにファイル名を突っ込むのは無意味にクラスのコードが肥大化

確かにそうですね。一方で、「コンストラクタの引数が多すぎて汚い」というのもよく指摘される部分で、それも合わせて修正できるかなとも思ったのですが、肥大化するのは確かです。引数の問題は別の方法で解決した方が良いですかね？

> 完全に同じ場所にはいてほしくない

僕の当初の想定では、下記のようなものを想定していましたが
```
Dynamics/Orbit/Orbit.cpp, hpp
Dynamics/Orbit/InitializeOrbit.cpp, hpp
```

これだと同じ場所にいるので適切でないということですよね。どのあたりが適切ですかね？ 例えば次のような感じですかね？
```
Dynamics/Orbit/Orbit.cpp, hpp
Dynamics/InitializeOrbit/InitializeOrbit.cpp, hpp
```

sksat, 2022-04-18:
あ，"完全に同じ場所"，はファイルのことを意図していました．なのでその想定で大丈夫です．

200km, 2022-04-18:
わかりました。では、上で書いた想定のようにファイルを異動させるという方針で修正してみます。

200km, 2022-04-19:
https://github.com/ut-issl/s2e-core/pull/91

I made a draft PR for this issue. Please check the direction of modification is fine.