ut-issl/s2e-core #13: Modify thermal dynamics

200km, 2022-01-05:
## Overview
Modify thermal dynamics

## Details
Currently, thermal dynamics calculation is implemented, but we did not confirm that it is working well.  
We need to confirm the implementation and fix it to work well.

## Conditions for close
When thermal dynamics is modified.

## Supplement
NA

## Note
- Write a conclusion when closing the issue.

200km, 2023-04-07:
## 進め方
- まずは現行のコードを @tshibuk に確認してもらい、ちゃんと動くか、どこが問題かを整理してもらう。
- コンポーネントの発熱量情報、搭載位置情報(つながっているノード情報)をS2Eの中でどう表現するか、熱計算と同インターフェースを取るかはもう少し考える ( @200km @sksat @tshibuk  )
  - 例えば、次のようなのがある
    - 全部熱計算に押し付ける
    - PowerPortのようにThermalPort的なのを作る
    - PowerPortに情報を付加する
  - これを考える時、「複雑な計算のためには、S2E外部ツールとの通信が必要になる」ということも踏まえて考えていくのが良さそう。

tshibuk, 2023-04-11:
現状のコードの整理
- 機器発熱量は初期値で一定しか指定できない
- 太陽入射量は単純な内積計算で算出

ひとまず修正したいところ（熱ソルバーとしての機能を入れるために）
- 時系列で各ノードの入熱量（機器発熱量と外部入熱量の和）を入力すると時系列の温度出力を出せるようにする
  - オプションで変えられるようにしたい
  - この時系列データはNodeオブジェクトやTemperatureオブジェクトの外で管理したい、なぜなら将来的にはこの時系列データなしでも（S2Eと連携して）発熱量を計算できるようにするつもりなので
    - HeatLoadオブジェクトみたいなのを作ることになりそう
- 時系列の入熱量の値から、線形補間でdtに合わせた入熱量を計算する機能を実装する
  - HeatLoadオブジェクト内で関数として定義するのがよさげ
- [Nodeの設定csv](https://github.com/ut-issl/s2e-core/blob/develop/data/sample/initialize_files/thermal_csv_files/node.csv)の形式はちょっと考えたいが、ひとまず変えないでおく

ここまで実装できれば渋川自作ソルバーと同等の機能を持つはず

tshibuk, 2023-04-12:
コードを眺めているとBoundaryノードに対する処理がなさそうなのでそこの追加は必須

tshibuk, 2023-04-13:
@200km 僕ってs2e-coreへのpushの権限って付与されてますか？pushしようとするとエラーが出るので…

closed 2023-05-22 by 200km