ut-issl/s2e-core #25: Use inertial frame data in SrpEnv

200km, 2022-01-05:
## Overview
Use inertial frame data in SrpEnv.

## Details
Currently, the body frame data is used in the class, but it is better to use the inertial frame data for simple implementation.

## Conditions for close
When the implementation is complete.

## Supplement
NA

## Note
- Write a conclusion when closing the issue.

suzuki-toshihir0, 2022-02-02:
@200km 対応遅れていてすみません．こちらですが，Shadow Functionの計算のときにbody座標系から見た地球・太陽位置が結局必要になるので，慣性座標系の量を使わなくてもいいように思いますが，どうする方針でしたっけ？（だいぶ前に軽く議論した気がしますが，どうなったか忘れてしまい．．．）

200km, 2022-02-02:
触に入るか入らないかは衛星の姿勢は関係なく、軌道だけの問題なので`body座標系から見た地球・太陽位置`は使う必要はないはずということだったと思います。

200km, 2022-02-02:
`衛星から見た` で良ければ先に差分など取ってr_sc2sun_eciベクトルとr_sc2earth_eci　ベクトルがあれば良いと思いますが、上の3つを与えて中で差分をとっても良いとは思います。  
地球触以外にも応用できるようにすることを考えると、どちらがいいかとかも考えて修正しても良いかもです。

suzuki-toshihir0, 2022-02-02:
わかりました．地球蝕だけ考える分には，どちらでも大差ないと思います．
今すぐ地球蝕以外に応用することは難しい（どの惑星による蝕を考えるかを与えるかのインターフェースを切る作業が要る為）ですが，パッと考えたIFの切り方だと

- SRPの初期化で，何の惑星の蝕を考えるかを与える
- SRPのクラス内部でそれぞれの惑星に対する蝕を考える

かなと思います．となると，SrpEnvの中には地球・太陽の情報だけでなくCelesInfoをまるごと入れてもいいかなとは思います．

200km, 2022-02-02:
SRP全体の話はそうですね。最初の時点でCelesInfoへアクセスできるようにしておいて、`UpdateAllStates`には衛星位置だけ入れるとかになるんですかね。  
いま`UpdateAllStates`に直で書かれているアルゴリズムは別関数に入れて、いろんな惑星毎に引数変えて使い回す的な感じでしょうか。

suzuki-toshihir0, 2022-02-02:
はい，流れとしてはそれが良いと思います．

suzuki-toshihir0, 2022-02-02:
もう一点，SRPクラスの中で，body座標系から見た太陽の方向をアップデートしていて，これをサンセンサやSAPが参照しているのが厄介ですね．これも別のところで定義しないといけなさそうです．

suzuki-toshihir0, 2022-02-02:
姿勢とは関係ないSRPの内部で`d_sc2sun_b_`を持っていて，それを更新しているのが変ですね．これは必要なくて，太陽方向を必要としているサンセンサとSAPにはlocal celes infoから太陽方向を提供すべきように思います．（そうしないと，UpdateAllStateの中からsun_position_bをなくせない）
作業内容としては，ここまで必要という感じですかね．

200km, 2022-02-02:
そうですね。body系太陽方向を計算・提供するのは別の場所でいいと思います。

200km, 2022-02-17:
実装されたのでCLOSEします。

closed 2022-02-17 by 200km