ut-issl/s2e-core #33: Decide Coding rules

200km, 2022-01-12:
## Overview
Decide Coding rules

## Details
The coding rule for S2E is described [here](https://github.com/ut-issl/s2e-documents/blob/main/General/CodingConvention.md), but the rule is unsophisticated.  
We need to change it better and more general for programmers.

## Conditions for close
When the coding rule is decided.

## Supplement
NA

## Note
NA

200km, 2022-01-12:
今までずっと放置してしまっていましたが、手を付けなければと思い、issueを立てました。

@sksat コード規約やスタイルについて特段強い意見は無いのですが、どういう風にしたら良いかアドバイスいただけると嬉しいです。

sksat, 2022-01-12:
C++ですし既存のスタイルに載っかってしまうのが楽ですね．`clang-format`とかでフォーマットできますし．
https://clang.llvm.org/docs/ClangFormatStyleOptions.html#configurable-format-style-options

200km, 2022-01-12:
ありがとうございます！その方針で行きたいと思うのですが、リンク先に書かれているLLVM,Googleなどからベースとなるスタイルを選ぶという感じでしょうか？おすすめはありますか？

@meltingrabbit C2Aは何をベースにしていましたっけ？できる限り近いのが良いかなと思っているのですが。

meltingrabbit, 2022-01-12:
clang-formatでGoogleルールを当てようとして，（C++ではなくpure Cなので）しんどいなぁとなって止まってます．

とはいえGoogleルールは定数などのルールが個人的にはあまり好きではない感じです．

ちょっとお待ち下さいね．

sksat, 2022-01-12:
C2Aと合わせた方がいいかもしれないというのはたしかにありそう．僕もあんまりちゃんとそれぞれのコーディングスタイルについてちゃんと把握できてないので確認してみてます．ただ個人的にはGoogleのスタイルはあんまり好きじゃないですね...(一番好みが分かれるスタイルでもある)．

meltingrabbit, 2022-01-12:
過去のメモみて思い出しました．

gnuは

[image]

が，個人的にはつらかったのでやめました．

LLVMはallman styleじゃなかった（C2Aは元来allmanなので）のでやめて，googleにした，という経緯です．

とはいえ，C2AはC++移行までは現状の  
https://github.com/ut-issl/c2a-core/blob/99b697d6733251f1599b561a03602892c5da115b/Script/CI/check_coding_rule.py  
でいくしかないかなぁとは思ってます．

そしてそれはそれなりに先のことなので，一旦C2Aを無視してもいいかもです．（なので，現状はgoogleに近くはあります）

meltingrabbit, 2022-01-12:
google styleでだいたい固めて，受け入れがたい部分は，個別設定（clang-formatの設定でできる）するというのもなしではないですが，その議論がめんどくさいというのもあります．

sksat, 2022-01-12:
GoogleのスタイルはあくまでC++のスタイルであってかなりC++に関するものが多い(そしてそこの好みがめちゃくちゃ分かれる)ので，S2EがC++である以上もうちょい慎重になりたいかもです．あとC2Aと合わせる云々はインデント関連とか命名規則とかぐらいでいい気もします．

meltingrabbit, 2022-01-12:
> あとC2Aと合わせる云々はインデント関連とか命名規則とかぐらいでいい気もします．

そうね．どうせ今後プロセス分離するなら，スタイルそのものを合わせる必要はないかも．（そもそも性質が違いすぎるので）

200km, 2022-01-12:
個人的にはC2Aとあまりに違うスタイルは両方一緒に書いているときとかに混乱してしまうので、C2Aと近いといいなとは思っています。  が、気になるのは主にはインデントとかの話なので、`C2Aと合わせる云々はインデント関連とか命名規則とかぐらい`でいいと思います。

200km, 2022-01-12:
その上で何かしらベースは決めないといけないのだろうと思いますが、どう決めていきましょうか。

meltingrabbit, 2022-01-12:
期限決めて各々スタイル眺めて，これがいいなぁと出し合うしかない気がします（好みなので）

C2Aに関しては，安全性の理由から修正してますが，こういうのをどうするかは別途議論してもいいかもです．

例
- https://github.com/ut-issl/c2a-core/blob/99b697d6733251f1599b561a03602892c5da115b/Docs/General/CodingRule.md#L447-L452
- https://github.com/ut-issl/c2a-core/blob/99b697d6733251f1599b561a03602892c5da115b/Docs/General/CodingRule.mdL474-L477

meltingrabbit, 2022-01-12:
あと，命名については，別途議論が必要かと思います．C2A踏襲してもいいですし，別途変えてもいいと思います．

特に，単位系などは，別で進んでるかもですが，型に押し込める方法で行きたいので，C2Aを踏襲したくない気持ちが強いです（C++なので）

200km, 2022-01-12:
はい。全て決めるの時間かかりそうなので少しずつ決めていきたいですね。
まずは、インデントなどから決めて、きれいにして行きたいなと思います。

命名の単位系は方に押し込める方針で良いと思いますが、それができるまではどうするかと言うのは決めたい気はしています。

sksat, 2022-01-12:
あとはとりあえず最低限欲しいものを上げていくのもありますね．個人的には
- `using namespace`のヘッダでの使用禁止(ヘッダ以外でも大きなスコープでは非推奨．特に`std`は．)
- 例外の使用禁止

は入れたいですね．

meltingrabbit, 2022-01-12:
上に加えて個人的に入れてほしいのは，

- インデントはスペースがいい（まあ4でいいのではと思う）
- メタコメント
  - https://github.com/ut-issl/c2a-core/blob/99b697d6733251f1599b561a03602892c5da115b/Docs/General/CodingRule.md?plain=1#L823-L827

ぐらいですかねぇ．

- CのenumではなくC++の使いましょうとか，適当な配列で頑張らずにSTLとかコンテナ使いましょうとか，そういう指針はあってもいいかもしれない？ とか？

C2Aの規約見てましたが，細かい話は，まず参照するスタイル選んでから出ないと議論できないので．

200km, 2022-01-12:
> インデントはスペースがいい  
たぶんこれは最近書いたのはスペース2つになっていると思います(現行ルールにもそう書いているので)  
名前がつくまえのむちゃくちゃ古いものや、外部から取ってきているものがそうなっていないという感じですかね。

他はOKです。

>まず参照するスタイル選んでから

meltingrabbit, 2022-01-12:
> インデント

ごめんなさい，ボケてました，2がいいです．（C2Aも2ですし）

meltingrabbit, 2022-01-12:
@sksat 参照するスタイル選び，どっかで時間決めてガット話します？

sksat, 2022-01-12:
> @sksat 参照するスタイル選び，どっかで時間決めてガット話します？

よさそう．それでお願いします．

200km, 2022-01-12:
時間合えば僕も参加しますが、優先度低めでいいです。

meltingrabbit, 2022-01-17:
- https://github.com/ut-issl/s2e-core/pull/35
- https://github.com/ut-issl/s2e-core/pull/36

をみて，low-level issueはどっちも同じなので，あまり関係はなく，あとはネーミングルールをどっちベースにするか，なきがしますね．

meltingrabbit, 2022-01-17:
LLVMのネーミングルールに手動で直すのめっちゃ大変そうなので，googleのフォーマットでいいのでは？という気持ちになってきた．

meltingrabbit, 2022-01-17:
C2Aの設定，一応貼っておきますね．

https://gist.github.com/meltingrabbit/40869b30a0863fa89fe392cf7840cd10

sksat, 2022-01-25:
ビルドCIも入りましたし，#35 をベースに`.clang-format`を拵えてフォーマットかけるやつどこかでガッとやってしまいます？

sksat, 2022-01-25:
clang-formatはネーミングルールまでは口田してこないようなので，少なくともclang-formatのベースはGoogleでよさそうです．ネーミングルールに関する強制(CIでチェックして落とすとか)は別の仕組みでやろうかな，と思っています．

200km, 2022-01-25:
分かりました！その方針で賛成です。

sksat, 2022-03-31:
1行の最大文字数を150文字に: #70

meltingrabbit, 2022-05-24:
@sksat これ実はcloseできる？

200km, 2022-05-24:
どこまでをcoding ruleと呼ぶかというのもありますが、今決まっているのはフォーマットだけで、命名規則などは決まっていないのかなと思っています。一度これをCLOSEして、別issue立ててもいいとも思います。

sksat, 2022-05-25:
一旦閉じてよさそう

200km, 2022-05-25:
では、CLOSEしましょうか。

closed 2022-05-25 by 200km