ut-issl/s2e-core #53: Decide Release rules

sksat, 2022-02-08:
## Overview
Decide rule about how to release/update S2E.

## Details
- [ ] versioning: (semver?)
- [ ] update rule
- [ ] document rule

## Conditions for close
When xxx is complete.

## Supplement
N/A

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

sksat, 2022-02-08:
バージョニングに関してはsemver(的運用)でよいと思います．C2Aもそうなりましたし(https://github.com/ut-issl/c2a-core/issues/141)．

sksat, 2022-02-08:
問題はどのようにバージョンを上げていくかなんですよね．もちろん「これは大きな更新だな」という時に(気分で)major versionを上げるとかも全然いいんですけど，S2E coreの後方互換性ってどう考えたらいいのか，という．

sksat, 2022-02-08:
C2Aのcore部もcoreというよりかはライブラリ的側面が強いけれど，S2Eはこれがもっと強い．どのくらい強いかというとS2E userは実態としてはS2E coreのforkという認識をした方がよい．

sksat, 2022-02-08:
で，本題なんですが，色々整理すると，「S2Eの互換性はcore/userで考えるべきではなく，**各コンポーネント**から**他のコンポーネント**とユーザコードに対して考えるべき」，というのが個人的な結論です．
文脈としては以下
- S2E userは実態としてはS2E coreのfork
- S2E coreは(現状はかなり密結合になっているが)各コンポーネントをまとめたmono-repoとして捉えるのがよい
  - このmonorepoには`src/S2E.cpp`などの **exampleとしてのユーザコード** も入っており，単体でビルドすることもできる

sksat, 2022-04-27:
あっ放置しててすみません．少なくとも形式上はsemver(`<major>-<minor>-<patch>`)でやる，ということで合意が取れていればよく，ドキュメントというかREADMEにそういう旨を書いておく，というぐらいですかね．あと既存バージョンのリリースをするならそれも．

200km, 2022-04-27:
> 少なくとも形式上はsemver(<major>-<minor>-<patch>)でやる

これで進めてOKだと思います！

> ドキュメントというかREADMEにそういう旨を書いておく

僕の方で書いていいと思いつつ、どんなふうに書けばよいかいまいち分かっていないので、やっていただけると嬉しいです。

> あと既存バージョンのリリースをする
これはあまり深く考えずに2.0~4.0をreleaseだけするという感じでいいですかね？

sksat, 2022-04-27:
了解です．
既存バージョンのリリースはやるだけで大丈夫です．ただしせっかくなので`2.0.0`ということにしてもいい(`2.0`とかはエイリアスということにして)かな，ぐらいの気持ちです．

200km, 2022-04-27:
わかりました。2.0.0にしておきます~

sksat, 2022-04-28:
既存のバージョンに対してpatch version付きのタグ追加 & GitHub Releaseでのリリースが行われた
- https://github.com/ut-issl/s2e-core/releases/tag/v2.0.0
- https://github.com/ut-issl/s2e-core/releases/tag/v3.0.0
- https://github.com/ut-issl/s2e-core/releases/tag/v4.0.0

sksat, 2022-05-11:
READMEにも書いたので閉じ

closed 2022-05-11 by sksat