ut-issl/s2e-core #56: STT throws up an error when the update step time of a component is greater than 1 second

suzuki-toshihir0, 2022-02-18:
## Overview
STT throws up an error when the update step time of a component is greater than 1 second.

## Details
### Description
When the update step time of the compo is large, the `output_delay` of STT is set to 1 in L.21 of `InitSTT.cpp`.
[image]

In this case, `MAX_DELAY` of STT is set to 1 as well.
[image]

As a result, the value of `hist` in `STT::measure` becomes -1, and referencing a negative index in `q_buffer` cause an error.
[image]

### Conditions of occurrence
```
CompoUpdateIntervalSec > 1.0
```
Need more detailed research.

###  Additional information
N/A

## Task
- [ ] Allow STT to work even when the update step of the compo is large

## Scope of influence
Analysis with large time increments (e.g., rough trajectory analysis) will be inconvenient.
In such a case, the user should first comment out the code related to the STT and deal with it.

## Supplement

## Note

200km, 2023-05-22:
@suzuki-toshihir0 これ、一応5月末までに解決したいなと思っているのですが、やれますか？やれないなら僕の方で引き取ろうかなと思っています。

suzuki-toshihir0, 2023-05-23:
すみません，お願いできますと嬉しいです．

200km, 2023-05-23:
この問題の本質は、`STT更新周期`が`STT出力遅延時間`より大きくなったこと。  
現状の設計ではそのような設定を想定していないので、`STT更新周期`>`STT出力遅延時間`の場合は遅延なしで出力すれば良い。（センサ情報を読み取るユーザーにとっては、最大パターンで`STT更新周期`分の遅延があるように見える）