ut-issl/s2e-core #58: Add calculation and logging flags to Attitude class

suzuki-toshihir0, 2022-02-23:
## Overview
Add calculation and logging flags to Attitude class

## Details
By turning off the attitude update, calculation with the attitude fixed at a specific quaternion `q_i2b` will be possible.
Implementing that feature will make it convenient to evaluate the magnitude of disturbances by fixing the attitude at various quaternions through Monte Carlo simulations.

## Conditions for close
calculation and logging flags implementation

## Supplement
N/A

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

200km, 2022-02-24:
If you want to fix the attitude, you can use the `ControlledAttitude` with the `INERTIAL_STABILIZE` mode.  
If you want to change the target quaternion for the `INERTIAL_STABILIZE` mode with the Monte-Carlo feature, it is better to modify the initialize interface for the target quaternion.

suzuki-toshihir0, 2022-02-24:
確かにINERTIAL_STABILIZEでいけますね。ありがとうございます。それとは別に、何故かソースコードの方でRK4での伝搬に計算するかどうかのフラグが入っていた（が、iniからはすでにフラグ設定は消されている）ので、それは消したほうがよさそうですね。

200km, 2022-02-25:
> 何故かソースコードの方でRK4での伝搬に計算するかどうかのフラグが入っていた

昔はControlledAttitudeがなかったので、このissueで提案してくれたようにフラグを持っていましたが、ControlledAttitudeができたので必要なくなったが、一応機能だけ残っているという感じですね。

ControlledAttitudeとモンテカルロをつなげるやつは、ControlledAttitudeに「初期姿勢のまま慣性指向」みたいなモードを作って、その初期姿勢を今すでにあるモンテカルロ機能でやるという感じでも良いかもしれません。

200km, 2022-04-25:
#108 で姿勢ダイナミクスを修正し、モンテカルロで初期Quaternionをランダムに振りつつ、シミュレーション終了まで固定値を維持することができるようになった。

対処方針はissueのタイトルと違うが、もともとやりたいことはできるようになったのでCLOSE。

closed 2022-04-25 by 200km