ut-issl/s2e-core #64: Bug in acceleration log

ShingoNishimoto, 2022-03-19:
## Overview
S2E cannot write the acceleration log correctly.

## Details
### Description
The acceleration term due to disturbances etc. is calculated in the acceleration of the Orbit class, but is reset for the next step in the Propagate function. Writing to the log is done outside of the Dynamics update, so the accelerations are reset to zero when they are written to the log.

### Conditions of occurrence
All environments.

###  Additional information
NA.

## Task
- [ ] Fix the error

## Scope of influence
We cannot check the spacecraft acceleration history from the log.

## Supplement
NA.

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

200km, 2022-03-19:
We can see the generated forces as the log data in the `Disturbance` class, for example, `airdrag_force_b` and `srp_force_b` etc.  
But I know that is only the short-term solution. We need to modify the place of the logging function in the `SampleCase.cpp`.

200km, 2022-04-19:
I will fix this issue since we need to quickly fix it before the next release.

200km, 2022-04-20:
@ShingoNishimoto このエラー修正されたと思います。ただし、ユーザー側の使い方も変わったので、これから書かれる新しいチュートリアルに沿って修正が必要です。

closed 2022-04-20 by 200km

ShingoNishimoto, 2022-04-21:
@200km ありがとうございます 🙏

ShingoNishimoto, 2022-06-04:
@200km I tried with v5.0.2 core, but there is no sat_acc_i_i left in the log (which is zero), so I think I am missing some necessary modification, can you please tell me the modification?

200km, 2022-06-04:
@ShingoNishimoto ユーザー側もドキュメントv 5.0.0に従いアップデートしたけれど、まだうまく行かないということでしょうか？

ShingoNishimoto, 2022-06-05:
@200km そうですね，ユーザー側を更新してcore v5.0.2で回るようにしたんですが，ログに残ってなかったので他に必要な修正があることだと思うんですが，それがぱっとわからないなという感じです．
ただ，coreで回すとログは残っているのでcoreのsampleケースを参考に見てみます．

200km, 2022-06-05:
[Tutorial](https://github.com/ut-issl/s2e-documents/blob/v5.0.0/Tutorials/HowToMakeNewSimulationScenario.md)にある通りの衛星、シミュレーションケースを作る必要があります。

ShingoNishimoto, 2022-07-23:
@200km 何度もすみません．こちらTutorialに従って作っているのにうまくログが残らなかったので，長い間放置していて，最近いろいろ調べてみたんですが，Orbit PropagationのモードがRK4だとうまくログが残ってないようです．試しにcoreにあるSampleCaseで試してみましたが，SGP4だとログが残るのに対してRK4だと全て0になってました．
自分の環境でログが残らなかったのもRK4を使っていたからで，SGP4にすると加速度ログが残っていることを確認できました．RK4で行う場合にログを残すためにはほかに必要な設定があるのでしょうか？

200km, 2022-07-23:
詳細報告ありがとうございます。  
RK4のUpdateの中で加速度を0にしてしまっていますね。修正PR出しておきます。

ShingoNishimoto, 2022-07-23:
ありがとうございます