ut-issl/s2e-core #70: Decide line length settings of clang-format

seki-hiro, 2022-03-28:
## Overview
Decide line length settings of clang-format

## Details
In Google C++ Style Guide, the [maximum line length](https://google.github.io/styleguide/cppguide.html#Line_Length) is 80 characters. In some situations, more than 80 characters may be desirable.

## Conditions for close
The line length settings of clang-format is decided.

## Supplement
[Related thread](https://github.com/ut-issl/s2e-core/pull/69#discussion_r834266183)

## Note
NA

200km, 2022-03-30:
@sksat [こちら](https://github.com/ut-issl/s2e-core/pull/69#discussion_r835733422)の議論にあるように、文字数制限を少し増やせたら嬉しいなと思っているのですが、いかがでしょう。

C2Aの方は130文字くらいまで許しているかと思います。

sksat, 2022-03-31:
返事が遅くなってすみません！1行の文字数制約は増やしていいと思います(あくまで統一されていることが大事なので)

200km, 2022-03-31:
ありがとうございます。では、文字数はC2Aと合わせようと思います。

sksat, 2022-03-31:
了解です．C2Aの方は100文字なんですかね？TBDとなってますが
https://github.com/ut-issl/c2a-core/blob/develop/Docs/General/coding_rule.md#%E6%94%B9%E8%A1%8C%E3%81%AA%E3%81%A9-w-f

200km, 2022-03-31:
https://github.com/ut-issl/c2a-core/blob/4805eb3e3f352365f255e8674479c14fa19ca1f8/Script/CI/check_coding_rule.py#L279

150文字見たいですね。

sksat, 2022-03-31:
お，なるほど

sksat, 2022-03-31:
とりあえず150文字でPR出しました

PR #71 (2022-03-31): Feature/source line length
## Overview
Resolve #70 

## Issue
- #70 

## Details
- update source file column limit(line length) to 150

##  Validation results
Link to tests or validation results.

## Scope of influence
eg. The behavior of XX will be change.

## Supplement
Write additional comments if you need.

## Note
- If there are related Projects, tie them together.
- Assignees should be set if possible.
- Reviewers should be set if possible.
- Set `priority` label if possible.
closed 2022-03-31 by sksat