ut-issl/s2e-core #85: 今後の方針を話し合うミーティングを行う

200km, 2022-04-17:
## Overview
今後の方針を話し合うミーティングを行う

## Details
- ミーティングは4/18 18時(JST)から
- 東大とAEのS2E主要メンバが参加する
- 議題は次の通り
  - 講習会に向けた改善点の洗い出し、優先順位付け
  - (講習会とは関係なく)追加実装したい機能の洗い出し、優先順位付け
  - 大規模改修の方針
  - 全体のスケジュール確認、マイルストーンの設定

## Conditions for close
- 今後の方針が決まったら

## Supplement
NA

## Note
NA

200km, 2022-04-18:
> バージョン番号

付け方は気持ちのによるが、次のことは守る
- 遡らない
- 何かしらの基準を決める

メジャー番号を上げるのは結構テキトウ。  
次は広い範囲の人に講習会をするという気概を込めた5.0.0にして、その後は細かくバージョンを上げていくというのが予想さそう。

sksat, 2022-04-18:
リリース・バージョニングに関する話: #53

200km, 2022-04-18:
> 5.0.0 にあげる前にこれだけは治しておきたいというポイント  
- 環境構築が大変。ライブラリを入れるところが困る。
  - ExtLibrariesについてはバッチファイルを使わずCMakeだけで良くなっている
  - ただし、ドキュメントは無いので書いたほうが良さそう 
  - VSでも動く？ -> VSからポチポチできる感じではない
  - この部分は一度まとめる
- #64
  - これは講習会前にやっとくので、新入生ではなく五十里がやる
- Initialize関係
  - Components.cppで直でインスタンス生成するような部分がハードルが高い
  - C++じゃなくて設定ファイルなどでできたら、、、
  - iniファイルを読む必要はない、コンストラクタが長いというのが分かりづらい原因な気がする
  - ここは別途考える
- ドキュメントの整理・充実
  - 講習会資料作成にあたって五十里の方でやっており
- iniファイルでの値設定周りが分かりづらい
  - TOMLフォーマットにするのが良さそう
  - iniに近いがまともなライブラリがあり使いやすい
  - これは講習会の後かな。。。

200km, 2022-04-18:
> 追加実装したい機能の洗い出し、優先順位付け  
- 対応する天体の拡張、地球周回以外も
  - ある程度できてるが、できていない部分も多分にあるので整理が必要
- 地上局の可視解析
  - もう少し追加実装すればできるようになるはず
  - これができれば通信系も使いやすくなる
- 結果の可視化
  - 現状はユーザーが好きに対応してくださいという形
  - 実行したらぽんとグラフが出てきてくれたほうがわかりやすい
- CI上でS2Eシミュレーションを回して結果を出す機能
  - PRを出したらコメントによく見るグラフが貼り付けられるみたいな感じ

200km, 2022-04-18:
> 大規模改修の方針  
- 近々でやろうとしていること
  - initComponent周りの整理
  - 単位系問題(単位を型にするなど)
- 東大側で手伝えること
  - ワーニング関連は東大でやれるかも
  - 細かいPRをいっぱい出すことも大切

sksat, 2022-04-18:
やった

closed 2022-04-18 by sksat