ut-issl/s2e-core #122: Release version 5.0.0

sksat, 2022-04-26:
## Overview
Major version update

## Details
Write in detail.

## Conditions for close
- [x] #85
- [x] #87 
- [ ] #53
- [ ] Release `v5.0.0`

## Supplement
Write additional comments if you need.

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

200km, 2022-05-02:
@sksat ドキュメントでv5.0.0をリファーしたいことが多々あるので、リリースしてしまってからドキュメントを修正したほうが良い気もしてきたのですが、どうでしょう？

200km, 2022-05-02:
それとも、URLは`https://github.com/ut-issl/s2e-core/releases/tag/v5.0.0`で確定だからそれを使えば良いですかね。

sksat, 2022-05-06:
すみませんメンション見てませんでした．とりあえずURLはそれで確定なのでPRを作っておいて同じようなタイミングでマージ，でよさそうです

200km, 2022-05-12:
リリースされたのでCLOSE!!

https://github.com/ut-issl/s2e-core/releases/tag/v5.0.0

closed 2022-05-12 by 200km