ut-issl/s2e-core #127: Improve updating time in SimTime

fujiy, 2022-05-09:
## Overview
Improve implementation of updating simulation time to not skip time for realtime simulation.

## Details

Relevant code:  [SimTime.cpp](https://github.com/ut-issl/s2e-core/blob/13d443f2806d983e2288290add1e5c8d0f0a9d56/src/Environment/Global/SimTime.cpp#L85)

Relevant discussion: https://arkedge-space.slack.com/archives/C02JRCY0A5V/p1646816850264289

### Cause

When running realtime (or faster, step_sec >= 1) simulation, normally your computer is fast enough.
But sometimes it needs more calculation than normal steps, then computing time may exceeds realtime.
In that case S2E assesses your computer performance is not enough and forcibly skips time.

### Problem
- S2E's time became faster and faster than C2A's time because [this code](https://github.com/ut-issl/s2e-core/blob/13d443f2806d983e2288290add1e5c8d0f0a9d56/src/Environment/Global/SimTime.cpp#L92) skips time, and we cannot properly compare S2E's logged value and C2A's telemetry value.
- Error message "Error: the specified step_sec is too small for this computer." is continuously displayed dispite your computer is fast enough.

### Possible improvement
Use average wait time over a short term (one second or something) instead of single step to check the computer is able to compute simulation steps in realtime.

## Conditions for close
When the implementation is complete or another solution is found.

## Supplement
NA

## Note
NA

fujiy, 2022-05-09:
https://github.com/ut-issl/s2e-core/blob/13d443f2806d983e2288290add1e5c8d0f0a9d56/src/Environment/Global/SimTime.cpp#L92
> カウントアップしてきた経過時刻を強制的に実際の経過時刻に合わせる
> 意図としては、ブレークポイントからの復帰後に一瞬でリアルタイムに追いつかせるため

この意図があまりわかっていないのですが，どんな風に使われていますか？
（自分で使うときはこの行をコメントアウトしています）

200km, 2022-05-09:
「ブレークポイントからの復帰後に一瞬でリアルタイムに追いつかせる」ということなので、デバッグ用に必要なものだと思いますが、古いコードで今となってはよくわからないという部分なので、必要性から考え直しても良いと思います。

closed 2022-05-24 by 200km