ut-issl/s2e-core #134: build error when USE_HILS ON

seki-hiro, 2022-05-17:
## Overview
build error when USE_HILS ON

## Details
### Description
build error
`コマンド ライン error D8016: コマンド ライン オプション '/clr' と '/MT' は同時に指定できません`

### Conditions of occurrence
USE_HILS is set to ON

###  Additional information
NA

## Task
- [ ] Fix compile option

## Scope of influence
HILS

## Supplement
NA

## Note
NA

seki-hiro, 2022-05-17:
@200km 
HILSで必要なc++/cliをコンパイルするための[/clr](https://docs.microsoft.com/ja-jp/cpp/build/reference/clr-common-language-runtime-compilation?view=msvc-170)オプションと/MTオプションとを[同時には使えない](https://docs.microsoft.com/ja-jp/cpp/build/reference/clr-restrictions?view=msvc-170)ようです。

common.cmakeの/MTオプションをコメントアウトするとビルドできるのですが、このオプションは何のためにあるか教えていただけないでしょうか？

200km, 2022-05-17:
HILSまで見てませんでしたね、、、warningを消すために追加したオプションですが、HILS ONのときは消したほうが良さそうですかね。

200km, 2022-05-17:
hotfix出す必要がありそうですね。

seki-hiro, 2022-05-17:
hotfixの方法がわからなかったのですが、ローカルではdevelopからしかブランチを切れないという理解であってますでしょうか？hotfixというのはmasterからの分岐を指しますか？

closed 2022-05-21 by 200km