ut-issl/s2e-core #135: Officially support 64bit build

200km, 2022-05-18:
## Overview
We need to officially support a 64bit build for s2e-core.

## Details
Currently, we recommend building the s2e-core as 32bit, but we can build it as 64bit without C2A.  
Several users ask us about the 64bit building. So we need to clarify the s2e-core can be built as 64bit without C2A.

TODO:
- CMake
  - [ ] Add 64bit build option (when USE_C2A = OFF)
- Documents
  - [ ] Add description of 64/32bit build
- CI
  - By the CMake modification above, 64bit building is executed when USE_C2A = OFF, and 32bit building is executed when USE_C2A = ON.

## Conditions for close
- TODOs are done.

## Supplement
NA

## Note
NA

200km, 2022-05-18:
@sksat 別の場所で議論されていたことに対応するために、このようなissue作って見ましたが、どうでしょうか。

sksat, 2022-05-18:
よいと思います！(まあほぼC2A側の問題ですけど)

200km, 2022-05-18:
はい。C2Aの問題であり、S2E単体は64bitでも32bitでもどちらでも良いのだというのを明確にするためにもこういうのを書いておけば、質問してくる人が減ったり、説明しやすいかなと思っています。とりあえずこの方針で修正かけてみます。

200km, 2022-05-24:
外部ライブラリのビルドも64bitに変更する必要があるので、少し面倒だと気づいた。

200km, 2022-05-24:
`USE_C2A = OFF`なら64bitはやめて、64bitにするかどうかのフラグをCMakeに持たせる方針が良さそう

sksat, 2022-05-25:
外部ライブラリ，むしろ自前でビルドしてる(ビルド済みバイナリ持ってきてるわけではない)のでそこはむしろ楽かも．とはいえS2EのCMake側でちゃんとケアしてあげる必要があるのはそうですね．

200km, 2023-05-15:
Windowsにも対応したのでCLOSE

closed 2023-05-15 by 200km