ut-issl/s2e-core #214: ExtLibraries build failed on Visual Studio 2022

sksat, 2022-10-28:
## Overview
SSIA

## Details
### Description
reported from @Yoshiki-Iwasa
error log:
```
ExtLibraries install dir: C:/Users/iwasa/SILS/S2E/s2e_core_oss/ExtLibraries/../../ExtLibraries/
downloading SpaceWeather-v1.2.txt
install nrlmsise to C:/Users/iwasa/SILS/S2E/s2e_core_oss/ExtLibraries/../../ExtLibraries//nrlmsise00
URLhttp://naif.jpl.nasa.gov/pub/naif/toolkit/C/PC_Windows_VisualC_32bit/packages/cspice.zip
CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.23/Modules/ExternalProject.cmake:2650 (message):
  URL_HASH is set to

    SHA256=bc566e6a975c888fc5fd89d76554329501446bda88c8bab937c0725faead170f;DOWNLOAD_EXTRACT_TIMESTAMP;false

  but must be ALGO=value where ALGO is

    MD5|SHA1|SHA224|SHA256|SHA384|SHA512|SHA3_224|SHA3_256|SHA3_384|SHA3_512

  and value is a hex string.
Call Stack (most recent call first):
  C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.23/Modules/ExternalProject.cmake:3640 (_ep_add_download_command)
  cspice/CMakeLists.txt:29 (ExternalProject_Add)
-- Configuring incomplete, errors occurred!
```

### Conditions of occurrence
- Visual Studio 2022
- CMake 3.23

###  Additional information
files, link, etc, if any.

## Task
- [ ] Fix xxx
- [ ] Do yyy

## Scope of influence
eg. All tools are not working.

## Supplement
Write additional comments if you need.

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

sksat, 2022-10-28:
問題が発生しているのはここ
https://github.com/ut-issl/s2e-core/blob/ec8d0aa15000e8e1067da07d5c5aede6e5914b1a/ExtLibraries/cspice/CMakeLists.txt#L29-L38

sksat, 2022-10-28:
`URL_HASH`がおかしくなってるのかと思ったけれど，本当の問題は直後の`DOWNLOAD_EXTRACT_TIMESTAMP`．これがCMake 3.24からの機能だった．
https://cmake.org/cmake/help/latest/module/ExternalProject.html
> DOWNLOAD_EXTRACT_TIMESTAMP <bool>
 New in version 3.24.
 When specified with a true value, the timestamps of the extracted files will match those in the archive. When false, the timestamps of the extracted files will reflect the time at which the extraction was performed. If the download URL changes, timestamps based off those in the archive can result in dependent targets not being rebuilt when they potentially should have been. Therefore, unless the file timestamps are significant to the project in some way, use a false value for this option. If DOWNLOAD_EXTRACT_TIMESTAMP is not given, the default is false. See policy [CMP0135](https://cmake.org/cmake/help/latest/policy/CMP0135.html#policy:CMP0135).

sksat, 2022-10-28:
対処案としては，
- `cmake_minimum_required`を3.24に上げる
- `DOWNLOAD_EXTRACT_TIMESTAMP`を使うのを一旦やめる

があり，できれば前者で行きたいものの， Visual Studio 2022でCMakeが3.23の環境が存在しているので難しい．

sksat, 2022-10-28:
https://cmake.org/cmake/help/latest/generator/Visual%20Studio%2017%202022.html
CMake Visual Studio 17 2022 Generatorが
> New in version 3.21.

とのことなので，3.21からもあり得るか

Yoshiki-Iwasa, 2022-10-28:
build後の `cmake --install` でなぜかCドライブ直下に `ExtLibraries`がインストールされる事象もありましたね

sksat, 2022-10-28:
あれは絶対パスを指定すべきやつでしたねたぶん(Windowsのパス周りは渋いので......)

Yoshiki-Iwasa, 2022-10-28:
ふむー
じゃあこのissueの内容とはまた違いそうですね

sksat, 2022-10-28:
ですね(まあいずれにしろドキュメントは書かないといけない)

KOBA789, 2022-12-30:
参考情報:
- jammy(22.04)は `3.22.1`
- kinetic(22.10)だと `3.24.2`