ut-issl/s2e-core #225: Modify all names to suit the current naming rule

200km, 2022-11-18:
## Overview
Modify all names to suit the current naming rule

## Details
Modify all names to suit the current naming rule

## Conditions for close
- all variable names are modified
- all function names are modified
- all parameter names in ini files are modified
- all log names in log file are modified

## Supplement
NA

## Note
NA

200km, 2022-11-27:
少し、[clang-tidy](https://clang.llvm.org/extra/clang-tidy/)を使ってみた。  
とりあえずこれを使えば自動チェックと修正ができるのでなんとかなりそうだが、次の部分をどうしようか悩んでいる。  
- 変数名や関数名に単位を入れている部分が今の[Naming Rule](https://github.com/ut-issl/s2e-documents/blob/develop/General/CodingConvention.md)と合わない。
  - ルールに明示していないが、わかりやすさのため新しいコードの多くで単位を明示している。大文字・小文字も通常使われる単位系に合わせているので、NamingRuleと合わない
  - [単位に関するissue](https://github.com/ut-issl/s2e-core/issues/223)を先に解決しなければ行けない気もする
  - 同様の問題は座標系でも言える。[座標系に関するissue](https://github.com/ut-issl/s2e-core/issues/224)を先に解決しなければいけない気もする
  - もしくはまずは単位や座標系以外の部分を修正し、後回しにするか。。。
- SGP4やIGRFなど自作じゃないコードにまで適用するかどうか
  - 基本は適用せずにそれらのファイルは無視するようにして良さそう

@sksat コメントいただけると嬉しいです。

200km, 2023-01-31:
@sksat こちら、ひとまずファイル名やディレクトリ名は人力で進めようと思っています。ただ、現在のファイル命名規則が `CamelCase`としているのに対し、将来導入したいのはGoogleスタイルなので、`snake_case`にすべきだなと思っています。このタイミングで全て`snake_case`にしてしまって良いでしょうか？

sksat, 2023-02-02:
コメント遅れてすみません......！！！

これって一旦大文字小文字だけ naming rule と合わせるとかでもとてもわかりにくくなってしまう（それぐらいだったらちゃんと修正したい），みたいなかんじですかね > 単位

はい．外部ライブラリは加味しなくていいと思います． > SGP4 や IGRF

このタイミングでいいと思います．十分に breaking change かつやるだけなのでやりましょう． > ファイル名を snake_case

200km, 2023-02-02:
ありがとうございます！

> 変数名や関数名に単位を入れている部分

これは、v6.0.0では単位と座標系以外は修正して、単位と座標系はv7.0.0以降に回そうと思っています。

> ファイル名を snake_case

これはv6.0.0でやろうと思います。documentの方の表記も変えておきます。

200km, 2023-02-08:
このissue対応していて気になったこと
- Interfaceディレクトリ配下のディレクトリ構成や命名を整理したい
- 各ディレクトリ配下のCMakeListについて整理したい（特にInterfaceや Libraryで細かく分かれすぎている件など）
- 名前として次のものは議論を呼びそう
  - Component/Abstract -> component/base_classed
  - SCIPort -> uart_port
  - Component/IdealComponent -> ?
- Component/Missionの下でさらにTelescopeディレクトリ必要？他と合わせるなら無くて良さそう

200km, 2023-02-08:
- ObcCommunicationBase などの命名方法なんとかしたい

200km, 2023-02-08:
- _tfsなどどうするか
- GSCalculator的な解析系をまとめるAnalysisディレクトリを作るか？

200km, 2023-02-08:
- Vector.hppがvector.hppになるのは、c++のvectorとの関係で嫌だったりするのか？

200km, 2023-02-08:
↑の件はそれぞれ別にissue作った
https://github.com/ut-issl/s2e-core/issues/312
https://github.com/ut-issl/s2e-core/issues/313
https://github.com/ut-issl/s2e-core/issues/314
https://github.com/ut-issl/s2e-core/issues/315

200km, 2023-02-09:
残存issueは別issueを作ったので、こちらはCLOSEする

sksat, 2023-02-13:
> * Vector.hppがvector.hppになるのは、c++のvectorとの関係で嫌だったりするのか？

閉じちゃった後でアレですが，これはそんなに嫌ではない（というか，その変更では"嫌さ"は変わらない）かなと思います．
"嫌"とすれば
- 人間側での conflict : 「ベクタ（クラス）」と言った時に指すものが２つある
- ソースコード上での conflict: （ファイル名ではなく）クラス名が `vector` だった場合，`using` すると区別がつかない & 最悪ビルド時に conflict してしまう

という理由になると思います

200km, 2023-02-24:
ファイル名は対応したが、クラス名や変数名はまだなので再オープンする。

200km, 2023-03-13:
Done

closed 2023-03-13 by 200km