ut-issl/s2e-core #238: Add attitude initializing feature with `ControlledAttitude`

200km, 2022-11-22:
## Overview
Add attitude initializing feature with `ControlledAttitude`

## Details
Currently, the initial quaternion is directly set by users. The feature to calculate the initial quaternion from the similar way with the `ControlledAttitude` will help the users to easy to set the initial attitude.

## Conditions for close
The feature is implemented.

## Supplement
NA

## Note
NA

kai0722, 2023-02-03:
おすすめはこのissueだったでしょうか？このissueに興味があります！

attitude initializing feature のイメージなど教えていただきたいです

200km, 2023-02-03:
@kai0722 ありがとうございます。では、こちらを是非お願いします。
このissueは、[この初期化ファイル](https://github.com/ut-issl/s2e-core/blob/develop/data/SampleSat/ini/SampleSat.ini)で指定している初期姿勢Quaternionである`Quaternion_i2b`をある条件に従って自動計算したいというものになります。

現在、姿勢計算には`RK4で積分伝搬する`モードであるRK4 と`理想的な姿勢制御を継続する`モードであるCONTROLLEDがあります。このissueは主にRK4に関係する機能追加なのですが、今RK4で姿勢計算をしていくときの初期姿勢Quaternionは、上記の`Quaternion_i2b`でユーザーが値を設定することになっています。

ただ、例えばRK4モードで`初期値として太陽電池が太陽方向を向いている方向にしたい`みたいなときに、この値をどんな値に設定すれば適切な姿勢になるのかというのは直感的にわかりづらく、別途計算が必要な状況です。この計算をS2E側でやってあげるようにしたいというのがこのissueで実現したい機能になります。  
ユーザーが`初期姿勢としてPX方向を太陽に向ける`みたいな設定を書き込めば、その姿勢を計算してくれて、そこから伝搬を始めてくれるという感じです。

「`初期姿勢としてPX方向を太陽に向ける`みたいな設定を書き込めば、その姿勢を計算してくれて」という部分は、実はCONTROLLEDモードでやっていることを最初の一回だけやれば良いということになり、そのコードを使い回すことができます。

200km, 2023-02-03:
以上をより具体的にまとめると、
- [SampleSat.ini](https://github.com/ut-issl/s2e-core/blob/develop/data/SampleSat/ini/SampleSat.ini)
  - RK4モードのときの初期姿勢Quaternionを今まで通りのユーザーが直接値を書くようにするか、CONTROLLEDモードで使っている`初期姿勢としてPX方向を太陽に向ける`という設定にするか選択するため、`initialize_mode`という設定値を追加する
- [InitAttitude.cpp](https://github.com/ut-issl/s2e-core/blob/develop/src/Dynamics/Attitude/InitAttitude.cpp)
  - この中で`initialize_mode`に従って、初期姿勢Quaternionを計算し、設定する

という感じになります。似たようなことは、[InitOrbit.cpp](https://github.com/ut-issl/s2e-core/blob/develop/src/Dynamics/Orbit/InitOrbit.cpp)で軌道に対してやっていたりします。(ただし、このコードは色んなものがグチャッているのでかなり見づらいとは思います)

kai0722, 2023-02-03:
ありがとうございます．色々見て取り組んでみます

200km, 2023-02-03:
正直、このAttitude周りのコードはすごく古いコードを微修正し続けている関係で汚くなっているので、色々難しい部分があるかも知れません。遠慮なく相談してください！

kai0722, 2023-02-08:
進捗（？）報告的なことなのですが、軌道力学のレポートが超大変で、手を付けるまでもう少しかかりそうです…

200km, 2023-02-09:
OKです！レポート頑張ってください！！

kai0722, 2023-02-26:
プルリクしてみたので確認お願いします！

closed 2023-04-04 by 200km