ut-issl/s2e-core #243: Clean up the directory structure in `Library`

200km, 2022-11-22:
## Overview
Clean up the directory structure in `Library`

## Details
`Library` directory structure is a mess.

## Conditions for close
- [ ] Rename all directories and files to suit with the naming rule
- [ ] `sgp4` and `RelativeOrbit` should go into the `orbit`
- [ ] Clean up the `math` directory
- [ ] Make environment directory to integrate `igrf` and `nrlmsise00`?
- [ ] Modify the index of s2e-documents after the cleaning

## Supplement
NA

## Note
NA

200km, 2023-02-08:
Ran0やRan1の名前をどうするか悩む。これは参考にしている教科書の関数名からとってきている。。。あと、Library外では参照されないはずなので、別の場所に移しても良い気もしている

200km, 2023-02-08:
次のように整理しても良さそう
- `quantization`は`library/utilities`に入れる
- 乱数系は`library/randomize`にまとめる
- 外部関数系は`library/external`にまとめる

いかがでしょうか？

suzuki-toshihir0, 2023-02-09:
フォルダの整理の仕方は問題なさそうです．

suzuki-toshihir0, 2023-02-09:
-  `Ran0` と `Ran1` については，別の場所（たとえば，`library/randomize/pseudo_random_generator` みたいな名前にする？ ）に入れるのが良い気がしますね．

- 名前は難しいですね．．．乱数の作り方でファイルが分けられていると，命名も乱数の作り方で決めざるを得なくなってしまう気がするので．
  - 一応，Ran0 は `minimal_standard_linear_congruential_generator` とかできなくはないかもしれないです．

200km, 2023-02-09:
Ran0 を `minimal_standard_linear_congruential_generator`にしたとすると、Ran1は`minimal_standard_linear_congruential_generator_with_shuffle`とかですかね。

200km, 2023-02-09:
linear_congruential_generator はlcgにしたい気もしますが、まあ名前長くてもよいか。。。

suzuki-toshihir0, 2023-02-09:
> `minimal_standard_linear_congruential_generator_with_shuffle`

そうなると思います．名前長いですが．．．

200km, 2023-02-09:
まあ、これで行ってみましょうかね。ちなみに、クラス名も`Ran0`ではなく`MinimalStandardLinearCongruentialGenerator`に変えることになると思います。もしくは、クラス名では、`MinimalStandardLcg`を許容するかどうかですね。

suzuki-toshihir0, 2023-02-09:
lcgと略すのは許容しても良い気はします．odeと比べて広く伝わるようなものではないですが，あまりにもクラス名が長いとコードが見にくくなってしまいそうなので．．．

200km, 2023-02-09:
とりあえずファイル名は長くても良さそうなので
`minimal_standard_linear_congruential_generator`  
`minimal_standard_linear_congruential_generator_with_shuffle`

で行ってみようと思います。クラス名の省略については別途議論しましょう。

suzuki-toshihir0, 2023-02-09:
了解です，一旦長いので行くのでよいと思います．

closed 2023-02-15 by 200km