ut-issl/s2e-core #247: Add ideal torque generator

200km, 2022-11-27:
## Overview
Add ideal torque generator

## Details
Add the ideal torque generator as a component.  
The [ForceGenerator](https://github.com/ut-issl/s2e-core/blob/develop/src/Component/IdealComponents/ForceGenerator.hpp) is a reference example.

## Conditions for close
- The component class is implemented and tested.

## Supplement
NA

## Note
NA

200km, 2022-12-23:
@kai0722 Detailsにも書いていますが、[ForceGenerator](https://github.com/ut-issl/s2e-core/blob/develop/src/Component/IdealComponents/ForceGenerator.hpp)のトルク版を作るというのが作業内容となります。ForceGeneratorをざっと眺めてみて質問があればまず質問してみてください。

kai0722, 2023-02-01:
ざっと見てみたところ，ForceGeneratorのForce部分をTorqueに変えれば良さそうに思えたのですが，Torque特有の変更点・追加点はありますか？

200km, 2023-02-01:
@kai0722 確認ありがとうございます。torque特有の部分は無い気はしています。もし実装していて出てきたら相談しましょう。

closed 2023-02-02 by 200km