ut-issl/s2e-core #258: Building the Docker image fails while trying to install gcc-multilib

shunichironomura, 2022-11-30:
## Overview
When building the Docker image per [this section](https://github.com/ut-issl/s2e-documents/blob/develop/General/HowToCompileWithUbuntuInDocker.md#32-make-docker-image-and-container) of the documentation, `./setup_docker.sh build` fails while trying to install `gcc-multilib`. Is this a bug or am I missing something?

## Details
### Description
When running `./setup_docker.sh build`, it fails with the following output:

```
 > [ 4/31] RUN apt-get update &&     apt-get install -y     cmake     gcc-multilib g++-multilib:                                                                                       
#7 1.097 Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease                                                                                                                    
#7 1.324 Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [114 kB]                                                                                                   
#7 2.764 Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [99.8 kB]                                                                                                
#7 3.220 Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [110 kB]                                                                                                  
#7 3.574 Get:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Packages [825 kB]
#7 4.312 Get:6 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [857 kB]
#7 4.555 Get:7 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse arm64 Packages [2335 B]
#7 4.560 Get:8 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [564 kB]
#7 4.691 Fetched 2572 kB in 4s (593 kB/s)
#7 4.691 Reading package lists...
#7 5.523 Reading package lists...
#7 6.162 Building dependency tree...
#7 6.286 Reading state information...
#7 6.320 Package gcc-multilib is not available, but is referred to by another package.
#7 6.320 This may mean that the package is missing, has been obsoleted, or
#7 6.320 is only available from another source
#7 6.320 
#7 6.486 E: Package 'gcc-multilib' has no installation candidate
#7 6.486 E: Unable to locate package g++-multilib
#7 6.486 E: Couldn't find any package by regex 'g++-multilib'
------
executor failed running [/bin/sh -c apt-get update &&     apt-get install -y     cmake     gcc-multilib g++-multilib]: exit code: 100
```

Changing the base image from `ubuntu:latest` (`ubuntu:jammy`) to `ubuntu:focal` did not resolve this issue.

### Conditions of occurrence

- Host machine: M1 Mac + Docker desktop
- commit: `602d24692e02b51c41d5af618b685b1adfef9714`

200km, 2022-11-30:
解決したようで良かったですが、この修正を反映すべきか考えたいのでreopenします。  
dockerfileを上手く書けばM1 Macでもそうでなくても動くようにできるのであればそうしたいです。  
最低限、documentの手順に注釈で入れたいかなと思っています。

sksat, 2022-12-01:
あー，これは正確には Docker image の aarch64 サポート事案っぽいですね．どうしたもんかな．

sksat, 2022-12-01:
まあ，そもそも `gcc-multilib` が必要なのが C2A との SILS のときに C2A を32bitビルドしたいことによって発生しているのがなんともではあるんですが

ItsukiKawamoto, 2023-05-13:
同様の環境で似たような箇所で詰まっています。https://github.com/ut-issl/s2e-core/issues/258#issuecomment-1332394429 に従い変更してみましたが通りませんでした。
`FROM --platform=linux/amd64 ubuntu:latest`ともしてみましたがダメなようです。
ログイン、ログアウト、アクセストークンでのログイン等、思い付くものは一通り試してあります。
また、Docker Engineの設定は`"buildkit": false`としてあります。
エラーメッセージは以下の通りです。
```
Dockerイメージを作成します
[+] Building 31.4s (4/4) FINISHED                                                                                                   
 => [internal] load build definition from Dockerfile                                                                           0.1s
 => => transferring dockerfile: 2.89kB                                                                                         0.0s
 => [internal] load .dockerignore                                                                                              0.0s
 => => transferring context: 2B                                                                                                0.0s
 => ERROR [internal] load metadata for docker.io/library/ubuntu:latest                                                        31.2s
 => [auth] library/ubuntu:pull token for registry-1.docker.io                                                                  0.0s
------
 > [internal] load metadata for docker.io/library/ubuntu:latest:
------
Dockerfile:1
--------------------
   1 | >>> FROM --platform=linux/x86_64 ubuntu:latest
   2 |     
   3 |     # ssh
--------------------
ERROR: failed to solve: failed to fetch oauth token: Post "https://auth.docker.io/token": dial tcp: lookup auth.docker.io: i/o timeout
```

200km, 2023-05-13:
@ItsukiKawamoto コメントありがとうございます！M1 Macユーザーの方に確認をお願いしていますので、少々お待ち下さい。

ItsukiKawamoto, 2023-05-18:
@shunichironomura 
上記コマンドを打った後リトライすると成功しました。ありがとうございます。

200km, 2023-05-18:
@ItsukiKawamoto 解決したようで良かったです！

@shunichironomura コメントありがとうございます！！