ut-issl/s2e-core #277: Fix RW ODE

200km, 2022-12-14:
## Overview
Fix RW ODE

## Details
Fix RW ODE adding time lag of the reaction wheel

## Conditions for close
Fix the function

## Supplement
NA

## Note
NA

LilKoke, 2023-02-12:
@200km このissueに興味があるのですが、これはどのようなissueでしょうか？

200km, 2023-02-12:
こちらは、指定されたトルクが実際に出力されるのに一次系で遅れて出力されるようにするというものです。  
昔実装していた形跡は残っているのですが、微妙だったので今はコメントアウトされています。一次遅れのパラメータは初期ファイルに入っていたりするので、それを使うことになります。

https://github.com/ut-issl/s2e-core/blob/develop/src/Component/AOCS/rw_ode.cpp

LilKoke, 2023-02-13:
なるほど
lag_coefが時定数で、出力はトルクではなくそのトルクに依存する角速度の微分なわけですね
衛星のダイナミクスをもう少し調べてこのissueに取り組んでもよいでしょうか？

200km, 2023-02-13:
ぜひお願いします！

200km, 2023-03-24:
@LilKoke 1次遅れ系ですが、下記のサイト等の運動方程式を参考にしてください。ここでの例は台車の移動になっていますが、回転体の運動方程式に置き換えれば同じです。  
https://controlabo.com/first-order-system-impulse-step-response/

今の実装だと、lag_coefficientsが3次元ベクトルになっていますが、これは昔の名残であまり意味はないので、必要に応じて次数を変えてくれて構いません。(もしくは2次、3次遅れ系にしても良いですが、実際あまり必要ないかなとは思います。)

わからないことがあれば遠慮なく質問してください。

closed 2023-10-11 by 200km