ut-issl/s2e-core #312: Clean up `interface` directory

200km, 2023-02-08:
## Overview
Clean up `interface` directory

## Details
`interface`ディレクトリの中身を整理する。特に`ILoggable`などの命名規則やディレクトリ配置。

## Conditions for close
NA

## Supplement
NA

## Note
NA

200km, 2023-02-08:
# このディレクトリの元々の意図
- ユーザーにとっての入出力インターフェースにあたる`initialize_file`や`log_output`に関連する物をおいておく
- S2Eとフライトソフトウェア(SILS想定)、S2EとPC外部(HILS想定)のインターフェースに関するものをおいておく
- ただし、`S2Eとフライトソフトウェア(SILS想定)`については微妙で、`power_port`などは各コンポとPCUの間の話でフライトソフトウェアとは直接は関わらない部分もある

# ディレクトリ名
- `Interface`とつけられていたのは、プログラミングにおけるインターフェースクラスという意味合いではなく、`S2Eとその外とのインターフェース`という意味である
- この中に`ILoggable`などがあるのは、たまたま

# インターフェースクラスの扱い
- 元々`インターフェースクラスはどこかのディレクトリに集めなければ`的な意識は無く、それぞれの場所にテキトウに置かれている。それが良いのか悪いのかもよくわかってはいない

# 議論したいこと
- interfaceディレクトリの中はv6.0.0アップデート提案のような感じで良いのか？
  - 特に、`sils`とした部分が実際には`sils`に関係しないもの(`power_port`)も無くはないことなど
- `ILoggable`のようなインターフェースクラスはどこに置くべきかと、その命名方法
  - `interface_loggable.hpp`でいいのか？`iloggable.hpp`なのか？

200km, 2023-02-09:
関連して気になる部分
- `com_port_interface`や`gpio_port`が`component/base`のなかのファイルを読んでいる

sksat, 2023-02-13:
`com_port_interface` とか，「S2E と S2E外とのインターフェース」という性質と「コンポの模擬」という両方の性質を備えているわけですよね．そして現状は前者を優先したディレクトリの切り方をしているわけですが，目的としては後者の方が強い（ので分割が歪になっている）と思います．

sksat, 2023-02-13:
なので，あくまで S2E と外界との界面の抽象化をするクラス（COM とか GPIO とか），HILS/SILS 用のコンポのクラス（これが前者に依存する），という分割をするのがよさそうです．

sksat, 2023-02-13:
`ILoggable` はかなり任意のクラスで使われていますし，そもそも意図がものすごく汎用のインターフェースなのでトップディレクトリ（`src/` 直下）でもいいのでは？という気もします．

sksat, 2023-02-13:
いや，これは`Logger`のインターフェースだし，`src/Logger` とかに`Interface/LogOutput`の中身を全部移すのがよさそうな気がするな

200km, 2023-02-13:
> src/Logger とかにInterface/LogOutputの中身を全部移す

これは確かにと思いつつ、`src/logger`と`src/interface/logger`の違いの重要性がいまいちわかっていません。

interfaceというディレクトリがなくなって、
```
- src
  - logger
  - initialize
  - sils
  - hils
```
みたいにするのが前提にあるという感じなのでしょうか。

下記のようなものにしないことのメリットを教えていただけるとより納得できる気がします。
```
- src
  - interface
    - logger
    - initialize
  - sils/hilsはどこか別で
```

200km, 2023-02-15:
@sksat 少し考えたのですが、`logger`などはlibraryにあるのが自然なように思ってきたのですがいかがでしょう？
同様に`com_port_interface`のようなコンポとは直接関係ないものもlibraryで良いように思ってきました。

`sils/hils`はcomponentの中に移して良い気がしています。

まとめると、
```
- src
  - library
    - logger
    - initialize
    - communication
      - com_port_interface
  - component
    - base
    - ports
      - power_port
      - sils_uart_port
      - hils_uart_port
```
というようなイメージですがどうでしょうか

200km, 2023-02-15:
@suzuki-toshihir0 ご意見いただけると嬉しいです。

suzuki-toshihir0, 2023-02-16:
## loggerとinitializeの位置について

議論追いました．

- `src` 直下に置いて `src/component` や `src/dynamics` などと並列に並べるよりは，`interface` もしくは `library` の下に押し込めた方が違和感が少ないように思います．
- そして，これらの置き場所は特段 `interface` に切り出す必要はないように見えます．なので，`library` の下に置くのが自然だというのには賛成です．

## SILSとHILSの位置について

- `component/port`の下に置いてもよいとは思います．
- ただ，`HILSはいろんな形態があり必ずしもコンポーネントで使われるだけではない` というのはあると思うので，`component` の下に置いてもよいと判断した理由だけ捕捉いただけるとより納得できると思います．

200km, 2023-02-16:
@suzuki-toshihir0 ありがとうございます。

> HILSはいろんな形態があり必ずしもコンポーネントで使われるだけではない というのはあると思うので，component の下に置いてもよいと判断した理由

- 大体はcomponentで使われる
- それ以外は、`library/communication`のCOM_PORT_IFを別途使う

で良いかなと思ってます。

suzuki-toshihir0, 2023-02-16:
ありがとうございます．

> それ以外は、library/communicationのCOM_PORT_IFを別途使う

というので了解です．それなら，僕は以下のまとめ方で問題ないように思います．
https://github.com/ut-issl/s2e-core/issues/312#issuecomment-1430959157

sksat, 2023-02-17:
元の `src/Library` 配下に外部のライブラリを持ってきたものが入っていたのが印象に残っていて，それらと logger みたいな S2E 内部のコンテキストで汎用なライブラリを並べるのはちょっとどうかなと思って言っていました． > logger をトップディレクトリに
それらは既に v6.0.0 のブランチでは `src/library/external` に切り出されていたので問題なさそう & `src/library` 配下に入れる方が自然ですね．

200km, 2023-02-17:
ありがとうございます。対応PRもマージしたのでCLOSEします。

closed 2023-02-17 by 200km