ut-issl/s2e-core #314: Rename obc_communication_base

200km, 2023-02-08:
## Overview
Rename obc_communication_base

## Details
`obc_communication_base`や`obc_i2c_target_communication_base`などの命名がいまいちな気がするので考え直したい。

## Conditions for close
NA

## Supplement
NA

## Note
NA

200km, 2023-02-08:
@seki-hiro @sksat @suzuki-toshihir0 こちらもなにかアイデアあればいただきたいです。  
- `obc_communication_base`は、OBCとUART通信するコンポが継承すべきクラス
- `obc_i2c_target_communication_base`は、OBCとI2C通信するコンポが継承すべきクラス
  - 基本はOBC側がControllerになるので、ここでは`i2c_target`となっている

という感じです。

suzuki-toshihir0, 2023-02-09:
### `obc_communication_base` について

- これはシリアル通信の基底クラスですよね．そのままの名前だとi2cなども含むように見えるので，`obc_serial_communication_base` などの方がいいかもしれませんね．
- `obc_` という接頭辞が何のためについているのかがよくわからず，かえってわかりにくくしているようにも思えます．このクラスを継承するのは（OBCクラスではなく）各種コンポ側なので，`obc_` という接頭辞はいらないのでは？とも思います．

suzuki-toshihir0, 2023-02-09:
### `obc_i2c_target_communication_base` について

- 上記の2点目と同様，`obc_`  という接頭辞はいらないのでは？という気がします．
- それ以外は現状変える必要は感じていないです．

200km, 2023-02-09:
コメントありがとうございます！

> obc_serial_communication_base

の用に具体性を高めるのは良いですよね。ただ、`uart_communication`のほうが適切かなとも思っています。(i2cも一応serial通信なはず？)

obcの接頭辞は、`obcと通信するための`という意味がこもっていると思います。
`uart_communication_with_obc_base`とかが丁寧な命名という感じですかね。

suzuki-toshihir0, 2023-02-09:
確かにI2Cもシリアルではありますね．．．であれば `uart_communication` が良いと思いますね．`uart_communication_with_obc_base` についても賛成です．

200km, 2023-02-09:
`obc_communication_base` -> `uart_communication_with_obc_base`は決まりで良さそうですね。  

I2Cについては、targetという単語をどこに入れるかが難しいんですよね。I2Cのcontroller/targetを明確にするために入れたいのですが、`i2c_target_communication`というのは英語としては微妙な気もしています。

あと、全てに`_base`を付けるべきかも議論したいです。他のファイルだと`spacecraft`とかもユーザーが継承する基底クラスなのですが、いちいち`_base`とかはつけないですし。

200km, 2023-02-09:
[こちらの議論](https://github.com/ut-issl/s2e-core/pull/311#discussion_r1101195469)にあるように、ファイル名の`_base`は消す方針で行こうと思います。

closed 2023-02-12 by 200km