ut-issl/s2e-core #316: Consider name of `ideal_components` directory

200km, 2023-02-08:
## Overview
Consider name of `ideal_components` directory

## Details
ideal_componentという名前が適切なのかは議論の余地あり。今のforce_generatorやtorque_generatorはaocsに入れても良さそう

## Conditions for close
NA

## Supplement
NA

## Note
NA

200km, 2023-02-08:
@sksat @seki-hiro @suzuki-toshihir0 こちらなにか意見ありますか？他のものが一応存在するコンポを念頭に置いている中、ForceGeneratorやTorqueGeneratorはそんな便利なものは世の中に存在しないけど、簡易的な制御実験に使えるという意味合いで用意しており、`ideal_components`としているのですが、なんだか微妙な気もしてきています。  
関連して、`gyro_sensor`よりもう一歩具体的な`特定の型番のgyro_sensor`みたいなのも今後追加されるとは思っていて、それをどこに配置するのかも同様に迷っています。`components/aocs`の中なのか、`real_component`とかにするのか。。。

suzuki-toshihir0, 2023-02-09:
### `ideal_components` について

- いまいちな点を整理すると，以下のような感じでしょうか．
  - `ideal_component` という命名から　`ForceGeneratorやTorqueGeneratorはそんな便利なものは世の中に存在しないけど、簡易的な制御実験に使えるという意味合いで用意しており、ideal_componentsとしている` というニュアンスが伝わりにくい
  - `ideal_component` という広い名前なのに，実際にはアクチュエータしか入っていない
-  一点目を考慮すると，これらのクラスは `example` の中に入っていてほしいような気もします．
  - `example` の中に，`structure` `communication` `actuator` みたいにフォルダを切って，`actuator` の中に `ForceGenerator` や `TorqueGenerator` を入れるイメージです．この場合idealという要素がなくなるので， `IdealForceGenerator` みたいな命名のほうがいいかもしれません．

suzuki-toshihir0, 2023-02-09:
### 特定の型番のコンポについて

これはcore側でケアすべきものなのでしょうか？特定の型番のコンポを入れたいとなってきたときはユーザーサイドでやるべきだと思っているのですが．．．

200km, 2023-02-09:
コメントありがとうございます。

`example`は、ユーザーが使うことは想定していないけど、ユーザーが独自コンポを作るときの参考になるためのコード　という風に捉えています。ForceGeneratorは解析主体のユーザーがガンガン使うことを想定しているので、ちょっと違うかなと思っています。

>ideal_component という広い名前なのに，実際にはアクチュエータしか入っていない

これに関しては、将来別のコンポを入れていけば良いと思うのでそこまで問題ではないと思っています。(例えば、今powerにあるpcu_initial_studyとかは移動させても良いかも)

> 特定の型番のコンポを入れたいとなってきたときはユーザーサイドでやるべき

これはユーザーサイドに限定する必要はないと思っていて、型番やICDが公開されていてよく使われるコンポの模擬コードを公開しておくと、ユーザーがそれぞれ独自で作る必要がなくなってより便利になると思っています。

200km, 2023-02-09:
- ideal_component的な場所を作るなら、名前をどうするか
- ideal_component的な場所は作らず、aocsなどの中に入れておくべきか

のどちらかを考えたいという感じになるんですかね。

suzuki-toshihir0, 2023-02-09:
ideal_componentに対する考え方のイメージがつきました，ありがとうございます．このイメージであれば

> 
> - ideal_component的な場所を作るなら、名前をどうするか
> - ideal_component的な場所は作らず、aocsなどの中に入れておくべきか

を考えていけばよさそうですね．

suzuki-toshihir0, 2023-02-09:
まず，`ideal_component` みたいな場所を作るべきかどうかですかね．僕の中のイメージだと，将来的にコンポーネントは

1. 解析用の理想的なコンポ（抽象度：大）
2. 今S2E COREに入っているような各種搭載機器の模擬コンポ（抽象度：中）
3. ICDが公開されている具体的なコンポ（抽象度：小）

みたいな感じに分類できるようになる気がしていて，これらを並列に扱うのが綺麗なのでは？という気もします．

suzuki-toshihir0, 2023-02-09:
そうなると，1. のために `ideal_component` みたいな場所は作るべきで，関連して `より具体的なコンポ` フォルダも作るべきということになるという感触です．

200km, 2023-02-09:
そうですね。3つを並列に扱うのがきれいだと思います。  
その場合、
```
- components
  - 解析用理想コンポ
  - 世に存在するコンポだけど、テレコマとかはないもの
  - ICDが公開されていて、テレコマなども模擬したコンポ
```
みたいに分けるべきですよね。

suzuki-toshihir0, 2023-02-09:
はい，分け方はそのとおりだと思います．

200km, 2023-02-09:
それぞれのディレクトリ名をどうするかが悩ましい。。。

suzuki-toshihir0, 2023-02-09:
バシッと綺麗な命名をあてるのは難しいですね．．．いくつか候補をあげるならば，

- 解析用理想コンポのディレクトリ
  - `quick_analysis`, `ideal`, `simple`
- 世に存在するコンポだけど、テレコマとかはないもののディレクトリ
  - `detailed_analysis`, `real`, `actual`, 
- ICDが公開されていて、テレコマなども模擬したコンポのディレクトリ
  - `specific`，`{世に存在するコンポだけど、テレコマとかはないもののディレクトリ} + with_communication`

とかですかね．．．`quick_analysis` とかはまあまあ意図を伝えられる命名な気はしますが，他がいまいちな気もする

200km, 2023-02-10:
センサはGyroだけど、アクチュエータはTorqueGenerator使うとかもありえて、その場合は一つのanalysisで`quick_analysis`と`detailed_analysis`が混ざるので、そういう意味ではanalysisの方法で分けるのは微妙だなとは思ったりもします。  
どんな解析に使うか、ある程度想定はしていますが最終的に決めるのはユーザーなので、そこをこちらが明示するような形にはしたくないですね。

型番のあるコンポは`products`が良さそうな気がしています。  
ほかは、ideal/realが対比もできて良さそうかなと思っています。

sksat, 2023-02-13:
- どの系のコンポなのか
- それがどんなコンポなのか
  - それが理想的なコンポなのか（これは"どんな"に含まれる）
- コンポの模擬のグラデーション（ex: 実在するならその型番など，テレコマまで模擬するのかどうか）

という順番に関心があると思っていて，それに即したディレクトリ構造・ファイル名を付けるべきだと思います．
上の議論見てパッと思いつくところでは
- components/
  - aocs/
    - ideal_hoge.cpp
    - gyro.cpp とか
    - products/
      - hogegyro.cpp とか？（具体的なコンポのクラスは Gyro.cpp とかを継承ないしメンバとして持って実装するのかなと思って書いてます）

みたいなかんじですかね．
ideal であるかどうかはコード上はコンポの性質のひとつでしかない（ディレクトリを切るほどではない）ように思います．特定の型番を付けるような実在のコンポについては，その性質上ファイル名の命名規則が他と異なることと，今後 core から分離できると便利かもしれない（core, user とは別に個々のコンポの模擬を共有できる場を分けられると便利な気がする），などからディレクトリを切りたいですね．
テレコマまで模擬しているかどうか，の部分をどうするかはちょっと考え中です．

200km, 2023-02-13:
idealなコンポーネントになると、解析の仕方によってかなり自由にコンポーネントが作られる（作りたい）と想定しているので、特定の系に分けづらいものもあると想像しています。  
例えば、絶対時間が分かるもの、蝕に入ったかどうか分かるもの、重心位置を変えるものなどがありえると思います。  ForceGeneratorもpropulsionかaocsかかなり微妙なところです。  

なので、idealで一つディレクトリを切った方がしっくりくるなと思っています。

また、`特定の型番がついている = テレコマが模擬されている`と想定しています。

sksat, 2023-02-14:
なるほどー．そこらへんの肌感はなかったです．
でも絶対時間や蝕に入ったかが分かるとかをコンポという名前でやるのはちょっと違和感ありますね．
「理想的なコンポ」と「シミュレータ - シミュレート対象（シナリオ・衛星の模擬・コンポの模擬）間の API」はちょっと違うもののような気がします．

sksat, 2023-02-14:
あー，でも結局それをどのように使いたいかによって結局「その API を使うコンポ」みたいなのは作りたくなりそうな気もするな

200km, 2023-02-14:
今のForceGeneratorなどと同様に「絶対時間を得る」みたいなコンポにもノイズをいれることを考えているので、真値系のAPIとは別ものだと思っています。（もちろんノイズを0にすることもできます）  ノイズを入れるというのがあるのでコンポーネントとして扱うのが自然と思っています。  
ノイズ無しで真値系のAPIから取ってくるようなシミュレーションもあり得ると思いますが、そういうシミュレーションは実用的なものでなくなってくると思っており、S2Eが目指す「研究したものをそのまま実用に回せる」というコンセプトからずれてしまうので、「理想的なコンポ」として定義することが重要だと思っています。

sksat, 2023-02-14:
ありがとうございます．ちゃんと納得できました．

200km, 2023-02-15:
@sksat @suzuki-toshihir0 こちらまとめると、次のような方針で良いでしょうか？

```
- src
  - components
    - ideal
      - ForceGeneratorなど主に解析で使う理想コンポ
    - real
      - acos
        - gyro_sensor
      - cdh
        - obc
      - 使い方はこれまでのcomponentsのような感じ
    - products
      - aocs
      - cdh
      - これまでのuser側の製品模擬の中で全体公開して良いものをコピーする
```

suzuki-toshihir0, 2023-02-15:
私はそれで問題ありません！

200km, 2023-02-16:
上の内容で進めようと思います。

sksat, 2023-02-17:
僕も問題ないです！

closed 2023-02-23 by 200km