ut-issl/s2e-core #317: 省略語の議論

200km, 2023-02-09:
## Overview
省略語の議論

## Details
今議論対象になっているのは次の部分
### ファイル名
- ode.hppは、ordinary_differential_equation.hppにしたが、`reaction_wheel_ode.hpp`は`_ode`のままでよいのか？
- 同様に`pcu_initial_study.hpp`の`pcu`はそのままでも良いのか？

### クラス名
- `ordinary_differential_equation.hpp`内で定義する`ODE`クラスは、`OrdinaryDifferentialEquation`クラスとすべきか、それとも`Ode`クラスでも良いのか
- 同様に`minimal_standard_linear_congruential_generator.hpp`内で定義するクラス名は`MinimalStandardLinearCongruentialGenerator`とすべきか、それとも`MinimalStandardLcg`でも良いか

## Conditions for close
When xxx is complete.

## Supplement
Write additional comments if you need.

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

suzuki-toshihir0, 2023-02-13:
省略語関連の議論が現状散逸しているように見えたので，まとめておく

- https://github.com/ut-issl/s2e-core/pull/311#discussion_r1101046412
  - [ ] ode.hppは、ordinary_differential_equation.hppにしたが、reaction_wheel_ode.hppは_odeのままでよいのか？
  - [ ] 同様にpcu_initial_study.hppのpcuはそのままでも良いのか？

- https://github.com/ut-issl/s2e-core/issues/243#issuecomment-1423568662
  - [ ] 同様にminimal_standard_linear_congruential_generator.hpp内で定義するクラス名はMinimalStandardLinearCongruentialGeneratorとすべきか、それともMinimalStandardLcgでも良いか
- https://github.com/ut-issl/s2e-core/pull/321#discussion_r1104387264
  - [ ] `implementation` を略して `impl` にしてもよいかどうか

200km, 2023-02-15:
The following abbreviations are available?
- calc
- init
-

200km, 2023-02-16:
なんとなく思ってることです
- 基本は略さない方針
  - すごく長くなる時、スコープが狭い時などは略してOK
- 1単語を短くするものは極力使わず明らかなものだけにしたい -> ホワイトリスト形式
  - `calc`, `init`, `max`, `min`などは、ほぼ一意に定まるのでOK
  - `impl`, `temp`などはあり得る単語が複数あるし、分かりづらいのでだめ
- 複数単語の頭文字を省略するのは、コメントなどで明確にわかるならOKにしたい
  - doxygenコメントにちゃんと書くならOKということ
  - ホワイトリスト形式にするのは大変なのでやりたくない
  - ただし、同じ略語になる危険性がある場合などはNG(reaction wheel, random walk)
  - ただし、略語のみはNGにしたい
    - `ode.hpp`, `rw_ode.hpp`はNG、`reaction_wheel_ode.hpp`はOK
- ファイル名とクラス名
  - 上のルールを守った上で、ファイル名は略してないけどクラス名は略すというのは合ってもいいと思う

suzuki-toshihir0, 2023-02-16:
### 基本のスタンス

- これは全面的に賛成です．

### 一単語を短くする方向の省略について

- ホワイトリスト形式にすべきというのも含めて，こちらも全面的に賛成です．

### 複数単語の頭文字をつなげて短くする方向の省略について

- コメントにちゃんと書くなら許容する，ホワイトリスト形式は大変なのでやらないというのは良さそうです．
- 同じ略語になるやつペアだけブラックリスト形式で洗い出しておく感じがいいですかね．
- 「略語のみはNG」というのはなんとなくそうかなとは思いますが，何故略語のみをNGにしたいかというロジックがあまりつかめてない感じもします．
  - たとえば， `obc.cpp` とかは許容してもよいような気もしてしまいます．

### ファイル名とクラス名

クラス名は必要に応じて略してもよいと思っています．

200km, 2023-02-16:
> 「略語のみはNG」というのはなんとなくそうかなとは思いますが，何故略語のみの命名をNGにしたいかというロジックがあまりつかめてない感じもします．

ここは感覚的なところが強く難しいのですが、`obc.cpp`まで短くしなくても良いだろうと思っているという感じですね。  
`on_board_computer.cpp`や`OnBoardComputer`でも実用上も全然問題ないなという気がしています。

ただ略語が重なってくると、それを全て書きだすのと全て略すのでは文字数が大きく変わり、どちらも極端な気がしています。
なので、`reaction_wheel_ode`のように重要な単語の方は書き出してしまうのが良いのかなと思っています。

suzuki-toshihir0, 2023-02-16:
そうですね．書いていて思いましたが，`on_board_computer.cpp` のほうがより望ましいと思います．
「書き下すと長すぎるけど，略そうとすると略語のみにならざるを得ない」みたいなケース（いい例がないですが．．．たとえば"Global Navigation Satellite System"はまあまあ長いが，略すとしたら"GNSS"と略語のみになってしまう）がないかだけ気になりましたが，正直S2Eで必要になるケースが思いつかないので大丈夫な気がしますね．

suzuki-toshihir0, 2023-02-16:
> 重要な単語の方は書き出してしまうのが良い

については，よいと思います．

sksat, 2023-02-17:
あまり略語を使うべきでないというのは（僕が見ていてよくわからないこともあって）とても同意するんですが，書き下したところで別にわかりやすくなるとは限らないので，全面的に略語のみは NG，というのは少し違和感があります（特にファイル名について）．ちょっとうまく言語化できてないので少しずつ書きます．

200km, 2023-02-17:
> 全面的に略語のみは NG，というのは少し違和感があります

難しいですが全面的にNGというか、
- まずは略さない方向で考えましょう
- 略したほうが良さそうな場合はレビューで議論しましょう

くらいが良いですかね。  
宇宙業界は略語が溢れていて、なんの疑問も無く略語使うことが多すぎて、これまでのS2EみたいにRWが被ったり、SSってなんなのとなったりしてきたので、方向性として略さない方向に傾けたい(でも略しても良いよ)って感じにしたいという意図ではあります。

200km, 2023-02-17:
まとめると

- 基本は略さない方針
  - すごく長くなる時、スコープが狭い時などは略してOK
  - OKかはレビューの過程で議論する
- 1単語を短くするものは極力使わず明らかなものだけにしたい -> ホワイトリスト形式
  - `calc`, `init`, `max`, `min`などは、ほぼ一意に定まるのでOK
  - `impl`, `temp`などはあり得る単語が複数あるし、分かりづらいのでだめ
- 複数単語の頭文字を省略するのは、コメントなどで明確にわかるならOKにしたい
  - doxygenコメントにちゃんと書くならOKということ
  - これをホワイトリスト形式にするのは大変なのでやりたくない
  - ただし、同じ略語になる危険性がある場合などはNG(reaction wheel, random walk)
  - 略語のみは基本はNGの方向で考える
    - `ode.hpp`, `rw_ode.hpp`はNG
    - `reaction_wheel_ode.hpp`はOK
    - 場合によってはOKになるのでレビューで議論する
- ファイル名とクラス名
  - 上のルールを守った上で、ファイル名は略してないけどクラス名は略すというのはあってもよい

sksat, 2023-02-17:
ありがとうございます．ちょっと思ったことを書いておきます．

略語（を使いすぎること）のデメリットを整理すると，
- 略語だけだと知らない人間が意味を把握できない
- 意味を知っていても，同時にたくさんの略語を把握しなければならないと認知負荷が上がる（脳内で巨大な変換テーブルを持たないといけない）

ということになると思うんですが，これの解決方法として
- 同時に把握しないといけない概念（略語を含む）の数を減らす = 適切にコンテキストを区切る
- 適切な（わからない時にアクセスしやすい）ドキュメントを用意する（ファイル中のコメントを含む）
  - ファイル名そのものについては本質的にドキュメントが別の場所にしか書けない
  - 近いところとして，そのファイルの中身（ヘッダとか），そのディレクトリの README とかを使うことはできる
- 略語を使わずに書き下す

などがあって，上から順に優先度が高いと思っています．

なので，たぶん一番同意できていないのは「ファイル名は略してないけどクラス名は略す」ですね．
- そのファイルの中身（クラス）に関心があるのであれば既に脳内の変換テーブルに載っているので冗長
  - （巨大でない）変換テーブルが既にある場合，略語は意味を圧縮するものというよりは固有名詞的にというか，ラベルとして扱われる
  - 略語をラベルとして扱っていると，むしろ逆変換にコストがかかる（ex: GNSS, S2E, C2A）
- 略語を知らない人間向けとしても，ファイル名を見て意味を把握 -> 中身を見て略語を把握，というのはあまり親切ではない
  - 変換テーブルを作るコストがかかるタイミングをほんの少し後回しにしただけに思える（そして逆変換のコストがじんわり増加する）
  - 単にファイル一覧を見た時のコストは下がる
  - ↑の２つのコストを考えた時に，両方略さない or 両方略すけど意味を明示する，のどちらかに寄せるべきなのでは？

みたいなことを思いました．

あと，これはファイル名ではなくクラス名とかの方ですが，S2E でものすごく認知負荷が高いものとしてものすごくたくさんの引数があるコンストラクタとかがあると思っていて，そういったところからとても長い１行みたいなのが発生しないかをちょっと心配してます．

200km, 2023-02-17:
> ファイル名は略してないけどクラス名は略す

こちらの意図は指摘してくださっている`ファイル名そのものについては本質的にドキュメントが別の場所にしか書けない`に対するものです。これがあるので、ファイル名自体は略さない方針で行きたいという思っています。  
一方で、クラス名は適切にコメントを残せばエディタの機能で表示されたりもすると思っているので、省略しても良いのかなと思っています。

> そういったところからとても長い１行みたいなのが発生しないか

これがあるので、クラス名をすべて書き出すというルールにはしたくないという思いもあり、上と合わせて`ファイル名は略してないけどクラス名は略すというのはあってもよい`という折衷案が良いと考えています。

> ファイル名を見て意味を把握 -> 中身を見て略語を把握，というのはあまり親切ではない

多分この辺りが意見の違ってくるところだと思うのですが、僕が思う体験は

- ファイル名を略語無しで見る
  - 略語だけ先に見えるというのが回避できる
- ファイルを開けてクラス名を見ると、略語と略語なしが同時に見える
  - 両方同時なので問題ない
- ファイル名見る前に、別ファイルから直接クラス名を見た場合は、略語と略語なしが同時に見える
  - 両方同時なので問題ない

みたいな感じだと思っているので、このやり方であれば、略語だけ先に見えて意味がわからなくてなにか検索をしようとするというのがなくなるのがメリットだと思っています。

> 略語をラベルとして扱っていると，むしろ逆変換にコストがかかる（ex: GNSS, S2E, C2A）

これが一番の問題だと思っていて、略語のほうが一般的になっているなら、それはそのままにするしかないかなと思います。(SGP4とかはスペルアウトしたほうがわかりづらくなりますしね。)  
実際、`gnss_receiver`や`gnss_satellite`が良いと思っているのは、`gnss`だけでなくその後に続く単語で意味が絞れると思っているからです。`c2a`も`onboard_computer_with_c2a`とかの文脈なら一意に定まるので使っていいと思います。略語のみにせず、他の言葉と合わせればかなりわかりやすくなるというのがあるので、`略語のみは基本はNG`としたいという感じです。基本NGとすることで、どういう単語と合わせればわかりやすいか考えてもらい、よりわかりやすい命名にしたいという意図もあります。
  
`s2e`は流石にこのプロジェクト全体の略語なので、OKとしたいところですね。

それ以外の略語について、どれが一般的かと言うのはやはり議論が必要になってきますが、それはレビューのたびに決めるしかないかなと思います。

200km, 2023-02-17:
話変わりますが、コンストラクタ長い問題は、一般的にはどう回避しているのでしょうか。  
こうすればよいというのがあれば、その方向に修正していきたいと思っていますが、わからないなという感じです。

sksat, 2023-02-17:
ありがとうございます．基本NG（レビュー必須）とすることで命名を考え直すきっかけを作るということであればとても賛成です．
宇宙業界が略語で溢れているという状況は本当にそうですし，方針に反対というわけではなく，こちらがどのように・何を心配しているかを明示したいというのが主な意図でした．

200km, 2023-02-17:
> 宇宙業界が略語で溢れている

本当にそうで、何も意識しなければ自分も略語使いまくっちゃいますし、レビュワーも略語について気にもとめず、スルーしちゃうんですよね。  
意識合わせはできたと思うので、後は運用の中で細かく詰めて行くということでお願いします。

sksat, 2023-02-17:
コンストラクタが長い問題は概ね引数が多すぎるということですね．
基本的にコンストラクタでは最低限のメモリ確保などの初期化などに徹するべきで，それ以上のことをするなら別のメンバ関数に切るべきだというのが （C++ の）セオリーだと思います．

他のクラスの生ポインタを大量に持っているクラスが多いのもあまりよくないと思っています．
その結果として，`クラスAのコンストラクタ(クラスBのポインタ, クラスCのポインタ, クラスDのポインタ)` みたいなコードを散見する気がします．
こっちは設計にも関わるのでなんともですが．

200km, 2023-02-17:
> 基本的にコンストラクタでは最低限のメモリ確保などの初期化などに徹するべきで，それ以上のことをするなら別のメンバ関数に切るべき

setの関数をたくさん作り、今ある`initialize_hoge`の中でsetしていくということですかね。今回やるかはさておき、そういう方針が良いということで理解しました。

> 他のクラスの生ポインタを大量に持っているクラスが多い

これはどう設計すればこれを回避できるのか、ミーティングなどで教えて下さい。

200km, 2023-02-25:
https://github.com/ut-issl/s2e-documents/pull/58

ルールも更新したのでCLOSE

closed 2023-02-25 by 200km