ut-issl/s2e-core #333: mathライブラリの微妙なところを修正する

200km, 2023-02-25:
## Overview
mathライブラリの微妙なところを修正する

## Details
> Matrix
- unitalize必要？最低限名前変えたい
- transposeやfill_upなどメンバー関数にしてはダメなのか？

> Vector
- normalizeを非破壊関数にしたい
- normなどをメンバー関数にしてはダメか？
- dotやcrossを使わなくするか？

## Conditions for close
When xxx is complete.

## Supplement
Write additional comments if you need.

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

200km, 2023-03-02:
@suzuki-toshihir0 @sksat 基本、上に書いている部分を修正していく方針にしたいのですが、懸念点などあれば教えて下さい。

sksat, 2023-03-02:
`normalize()` を非破壊関数にしたいというのは，あるベクトルの変数を正規化したものを使いたいときに元の値も使いたい/使い続けたいことが多いということですかね？

sksat, 2023-03-02:
`Matrix` の `transpose()` とか `Vector` の `norm` などをメンバ関数にするのはとても賛成です．複数の `Matrix` や `Vector` を引数に取るとかでなければできるだけメンバ関数に寄せたいです．

200km, 2023-03-02:
> あるベクトルの変数を正規化したものを使いたいときに元の値も使いたい/使い続けたいことが多い

そうですね。あとは、ここで元の値が書き換わることに気づかず使われていて計算が間違っている例などが過去にありました。  

S2Eで破壊される関数がほぼ無い中、Vectorのこの関数だけ破壊されるのが罠だと思っています。

200km, 2023-03-07:
あと、`matrix_vector.hpp`にある`Inverse`や`LU分解`などもすべて`matrix.hpp`に移そうと思います。

closed 2023-03-19 by 200km