ut-issl/s2e-core #381: Add initial angular velocity for reaction wheel

suzuki-toshihir0, 2023-04-05:
## Overview
Allows the initial angular velocity of reaction wheels to be specified in the ini file.

## Details
It is expected to be useful for attitude analysis without using C2A and simulations to evaluate attitude instability by RW jitter.

## Conditions for close
- [ ] Add initial angular velocity for reaction wheel in ini file

## Supplement
NA

## Note
- If there are related Projects, tie them together.
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

suzuki-toshihir0, 2023-04-05:
@200km RWの初期角速度をiniファイルで設定したいなと思っています．姿勢解析でこういう機能があると便利だと思うことがたまにあるので，特に違和感がなさそうであればさっと進めたいと思っていますが，どうでしょう？

200km, 2023-04-06:
一応初期化ファイルには存在していますが、中では使われていないという状況なので、ちゃんと使えるようにしたいということで良いでしょうか？  
https://github.com/ut-issl/s2e-core/blob/feature/major-update-v6.0.0/data/sample/initialize_files/components/reaction_wheel.ini#L50

suzuki-toshihir0, 2023-04-06:
あ，すみません，あることにはあるのか．．．

200km, 2023-04-06:
はい。すでに書いているように、`中では使われていないという状況`です。

suzuki-toshihir0, 2023-04-06:
ですよね．中で使えるように改修はしたいので，それは進めようと思います．

200km, 2023-05-22:
@suzuki-toshihir0 こちらは次のminor updateに回すということで良いでしょうか？

suzuki-toshihir0, 2023-05-23:
次回のminor updateでお願いします．今回のに間に合わせるのは手が回らないので．．．

closed 2023-10-02 by 200km