ut-issl/s2e-core #392: Refactor I2C communication interface

200km, 2023-04-14:
## Overview
Refactor I2C communication interface

## Details
Currently, I2C communication interface has many limitations. We need to improve the class for realistic behavior.
- Register size is limited as 1 byte for 1 address.

## Conditions for close
- I2C is refactored

## Supplement
NA

## Note
NA

TomokiMochizuki, 2024-12-23:
There is a component that sends multiple commands via i2c in one cycle of c2a.

200km, 2024-12-23:
> ミーティングで話し合った方針
- I2Cポートにオーバーライド可能な自由度の高いコマンド処理関数を作る
- コマンドが単純なレジスタアドレスの形式でない場合は、ユーザーでそのコマンド処理関数をオーバーライドして任意の処理が書ける様にする。
- 上の基本方針を`i2c_port`と`i2c_target_communication_with_obc`両方をうまく修正することで達成する
- 具体的な部分はもう少し整理する。 -> @200km 
- 整理できたら、 @TomokiMochizuki や @sksat さんにも説明してから実装に入る