ut-issl/s2e-core #405: plot_satellite_orbit_on_miller,py might have bug

DaikiTsuzuki, 2023-05-10:
## Overview
plot_satellite_orbit_on_miller,py might have bug

## Details
### Description
satellite orbit is poltted like this
[image]

The right and left ends are connected by a line.

### Conditions of occurrence
The hash in which the bug occurs is `7fc6df46`, and the log file is in [https://drive.google.com/drive/folders/1I6muwtwqflZkFgqWR3HTmSq4Mahy_4V5?usp=sharing]

###  Additional information

## Task

## Scope of influence

## Supplement

## Note

200km, 2023-05-10:
@DaikiTsuzuki バグ報告ありがとうございます。
私の環境でも確認できました。  

[この資料](https://github.com/ut-issl/s2e-documents/blob/develop/General/HowToVisualizeSimulationResults.md)を作った時点ではこのバグはでていないようです。その後のこのscript関連の修正は、matplotlibなどのライブラリのバージョン更新だけなのでその辺りが怪しいかなと思っています。

200km, 2023-05-10:
linestyle=’None’　設定すれば解決する気はする。

DaikiTsuzuki, 2023-05-12:
解決しました。ありがとうございます。

200km, 2023-05-12:
こちら、元コードの方も `linestyle=’None’`を入れるように修正しようと思うので、Reopenしました。

簡単な修正なので、研究室の新入メンバーにやってもらおうと思っているので少々お待ち下さい。

200km, 2023-05-19:
こののバグが発生するのは、次の2つ
- [plot_gs_visibility.py](https://github.com/ut-issl/s2e-core/blob/5edc3e5b8ad25477a8de06cdb503b52460440a52/scripts/Plot/plot_gs_visibility.py)
- [plot_satellite_orbit_on_miller.py](https://github.com/ut-issl/s2e-core/blob/5edc3e5b8ad25477a8de06cdb503b52460440a52/scripts/Plot/plot_satellite_orbit_on_miller.py)

それぞれの`map.plot(sc_map_lon[i], sc_map_lat[i], color='blue', marker='o', markersize=3)`などに`linestyle='None'`設定を入れるだけでと思っています。
担当割りは
- plot_gs_visibility.py -> @ogoogo 
- plot_satellite_orbit_on_miller.py -> @Hiro-0110 
でお願いします。

closed 2023-05-24 by Hiro-0110