ut-issl/s2e-core #423: Discuss remove library directory

200km, 2023-06-02:
## Overview
Discuss remove `library` directory

## Details
Please see the following discussion.  
https://github.com/ut-issl/s2e-core/pull/415#discussion_r1214405032

## Conditions for close
- Modification policy is decided.

## Supplement
NA

## Note
NA

suzuki-toshihir0, 2023-06-03:
### libraryというリリース単位が適切か？

`Library` という単位は大きすぎる，かついろいろなものが混ざっているのでまとめて扱わないほうがよいだろいう，という意見には同意です．

### src/ 直下に numerical_integration のようなディレクトリ/名前空間を直接切っていく方がよいか？

`Library` という大きすぎる単位でディレクトリ/名前空間を切ると不自然なので，ある程度細かく役割でグルーピングされた単位でディレクトリ/名前空間を切るべきだとは私も思います．いっぽう，この作業をやっていくにあたり，リリース単位の粒度は揃えておきたいなと思います．`math` と `numerical_integration` が並列にあるのは変で，`math` ⊃ `numerical_integration` だと思いますので．

### 名前空間の命名規則について

`library_math` という名前空間名については，もはや `library_` というprefixに意味がほとんどないと思っています．`math` そのものが，なにかしらのライブラリであるという意味を含んでいるように思えるからだと思います．その意味では，単純に `math` みたいな名前のほうがふさわしいと考えています．

200km, 2023-06-05:
コメントありがとうございます。

> libraryが必要かどうか

`library`がなくなった場合、src配下に`components`や`dynamics`と並んで、`math`、`communication`、`logger`、`optics`などが並んでいくことになると思いますが、分割粒度が合っていない気がしています。`components`などもなくして細かい単位をsrc配下に配置していくと言う感じでしょうか？  

libraryという命名は変えても良いですが、一旦`math`などを管理するディレクトリがあるのは見やすさという観点では分かりやすいと思います。
 
> `大きすぎる単位でディレクトリ/名前空間を切るのは不自然`

ということなので、同様に大きな単位である`components`や`dynamics`でディレクトリを切るのも不自然ということなのかもしれませんね。  
私の考えでは、ディレクトリは大きな分類でまず分けて、その後その中でさらに細かく分けていくのが良いと思っていますので、その最初のところが違うのかもしれませんね。

> mathの立ち位置
  
mathというのも、数学一般という意味の広義のmathと、数式関数ライブラリという意味の狭義のmathと捉えるかでも考え方が違うかもですね。前者にしてしまうと、かなり多くのものがmathに含まれてしまって、結局mathがかなり肥大化する気がして微妙だなと思います。C言語の`math.h`見た時に数値積分なども含んでいるとは思わないように、狭義のmathという風に普通は認識しているのだと思います。  
また、`math`が肥大化すると、`大きすぎる単位でディレクトリ/名前空間を切るのは不自然`というのが`math`に適応されることになり、`vector`, `matrix`, `quaternion`などで分けるのがよく、src直下にこれらをおくべきということになりませんか？

> 命名規則

今の方針だと、`library_math`ではなく、`libra::math`ですね。(略語を無くすために`library::math`にしても良いとは思います)

`library`という分け方が変なのでそれを消すということなら、そこでlibraryを残すのは変なので普通に`math`になると思います。なので、これは`libraryディレクトリを残すか`という話に帰着するものだと思います。

suzuki-toshihir0, 2023-06-05:
> libraryがなくなった場合、src配下にcomponentsやdynamicsと並んで、math、communication、logger、opticsなどが並んでいくことになると思いますが、分割粒度が合っていない気がしています。componentsなどもなくして細かい単位をsrc配下に配置していくと言う感じでしょうか？

意図がきちんと伝えきれてませんでした． `library` はいろいろなものが混ざりすぎて，なんのライブラリかわからなくなるので大きすぎるということが言いたいことでした．そして，分割粒度が合っていない問題については，現状でもlibraryの中身の粒度が揃っていないということが根本的な問題だと考えます．libraryをなくすなら，libraryの中身をある程度粒度を揃えるように整理することが必要になると考えています（このあたりの目線が @sksat と合っているかは気になる ）．

逆に，僕は 

> 同様に大きな単位であるcomponentsやdynamicsでディレクトリを切るのも不自然

とは考えていないです．そのディレクトリの中身がS2Eに対してどのような機能を提供するかが明らかな分類と命名になっているからだと思います．

> また、mathが肥大化すると、大きすぎる単位でディレクトリ/名前空間を切るのは不自然というのがmathに適応されることになり、vector, matrix, quaternionなどで分けるのがよく、src直下にこれらをおくべきということになりませんか？

これも，少なくとも `math` が数学関数を提供するライブラリであることがはっきりするディレクトリ・フォルダなのでその中がある程度肥大しても良いかなと思っていましたが，いっぽうで数値積分（や，他には乱数なども？）がmathに含まれるかは怪しいですね．このあたりを考え出すと，ディレクトリの分け方に個人の感覚が入り込んできそうなので，うまく議論がまとまるか心配になってきました．．．

sksat, 2023-06-06:
分割の粒度が合っていないというよりは，分割の目的が合っていない（ので並列すると思いにくい）のかなと思っています．S2E の中の1つのコンポーネントなのか，必ずしも S2E に限定したものではなく（実際のユーザが S2E に限定されてるとはいえ）ある程度汎用的に使い回せるような「ライブラリ」なのか，の違いというか．

例えば，`library` というディレクトリを切るのではなく（これは `library` という confusing な名前空間を無くすため），`src` の直下に `lib` prefix を付けたディレクトリを各ライブラリ毎に切るのはどうでしょうか（`src/libmath`, `src/liborbit` みたいな）．

`math` ライブラリも分割すべきなのではないかという議論も当然あると思っています．ただ，それはそれとして，この分割の主目的は「中身が肥大化している」からというよりは，（ライブラリとして）再利用したい単位として不自然だから，だと思っています（なので別の議論だと思います）．
（現状のディレクトリ構成・設計にちゃんと即すのであれば `library::math` のような名前空間を採用すべきだが）`library` という名前空間そのものを使いたいことは無いけれど，`library::math` に相当する名前空間を用いて「ライブラリ」を使いたいことはあって，であれば `library` 名前空間は要らないのでは（`library` という語はほぼ何も示していないようなものだし），というか

200km, 2023-06-07:
@suzuki-toshihir0 @sksat コメントありがとうございます。

@suzuki-toshihir0　が書いてくれた`大きすぎる単位`という表現は本質的ではなく、分類が適切かということが重要なのだと理解しました。

> library はいろいろなものが混ざりすぎて，なんのライブラリかわからなくなる

現状は、「S2Eの様々な場所で適宜使われる便利なコードで、他のコードとの依存関係が少ない小さなコードのの集合」という分類ですよね。そういう意味では、`ディレクトリの中身がS2Eに対してどのような機能を提供するかが明らかな分類`だと僕自身は感じています。命名は変えても良いとは思っています。この部分でも`個人の感覚が入り込んできそうなので，うまく議論がまとまるか心配`な気はしています。

> 必ずしも S2E に限定したものではなく（実際のユーザが S2E に限定されてるとはいえ）ある程度汎用的に使い回せるような「ライブラリ」なのか

このコメントや、上の`分類として適切か`　を考えるときに、`他のコードとの依存関係が少ないもの`という部分が重要な気がしますが、現状意識しているのは、`libraryディレクトリの中のものは、libraryディレクトリの中のものだけに依存して作られている`という部分です。例えば、
- `Vector`、`OrbitalElements`はC++標準機能にのみ依存している
- `KeplerOrbit`は`Vector`と`OrbitalElements`に依存している
- `Quaternion`は`Vector`に依存している
- `NumericalIntegration`は`Vector`と`Matrix`に依存している

のような状況です。`orbit`ディレクトリだけを抜き出しても、他では使えず`math`も必要になります。そのようなことから、`再利用したい単位`のようなことを考えても、`library`ディレクトリがあり、その下に適切に`math`や`orbit`が分割されている　というのは自然なように思っています。`initialize`や`logger`など数学・物理と関係ないものがあるのが違和感なのでしたら、下記のようにlibraryの中で更に整理しても良い気がします。これで`S2Eに対してどのような機能を提供するかが明らかな分類と命名`は達成できる気がしていて、`library`というレイヤーを消すかどうかではない気がしています。
```
- library (名前は変えても良いかも)
  - math_physics (名前はイマイチな気がするので変えたい)
    - math
    - orbit
    - numerical_integration
    - randomization
    - optics
    - geodesy
  - artificial (名前はイマイチな気がするので変えたい)
    - communication
    - initialize
    - logger
    - utils
  - external
```

> これは library という confusing な名前空間を無くすため

これは、僕が何に対してconfusingなのか理解できていないので、具体例を教えていただけないでしょうか？名前の問題だけなら、別の名前でも良いとは思っています。

もしかすると、「`library`の中でも`math`だけほしいのに、`library`全部持ってくると必要ない`orbit`も入るのがおかしい」ということを気にされているのでしょうか。それは、「`dynamics`のなかでも`orbit`だけほしいのに、必要ない`thermal`もはいるのがおかしい」と同じような気がして、その結果`dynamics`を消して`dynamics_orbit`を作っていくか　ということになったりする気もしています。  
「dynamicsは他の依存関係が大きすぎるから、それだけを抜き出すことは無いが、libraryは依存関係が小さいの抜き出す可能性があるから別議論」という感じなのかもしれません。

このあたりどのような考え方か教えていただけると嬉しいです。

200km, 2023-12-22:
# ミーティング議論の結果
次のようにsrc配下に依存関係が明確になるようにディレクトリ配置した方が良い。

```
- math_physics (名前はイマイチな気がするので変えたい)
  - math
  - orbit
  - numerical_integration
  - randomization
  - optics
  - geodesy
- external
*以下はもう少し整理した上で決める
- communication -> 特にこれどうする？
- initialize
- logger
- utils
```

200km, 2024-02-15:
> 現在の実装状況

- math_physics -> 命名議論したい
  - math
  - orbit
  - numerical_integration
  - randomization
  - optics
  - geodesy
- external_libraries -> 議論中
  - igrf
  - sgp4
  - など
- setting_file_reader
  - ini_access
  - wings_command_interface
  - など
- logger
  - CSVログ出力関連
- utilities
  - slip
  - endian
  - など

200km, 2024-02-21:
> 各種PRレビュー後、ミーティングでの議論後の最新状況

- math_physics -> この命名でGoが出た
  - math
  - orbit
    - sgp4
  - numerical_integration
  - randomization
  - optics
  - geodesy
  - geomagnetic
    - igrf
  - atmosphere
    - nrlmsise_wrapper
- external_libraries -> 削除
- setting_file_reader -> review待ち
  - ini_access
  - wings_command_interface
  - など
- logger -> Approve済み
  - CSVログ出力関連
- utilities -> Approve済み
  - slip
  - endian
  - など

200km, 2024-04-11:
[Remove library directory #621](https://github.com/ut-issl/s2e-core/pull/621)がマージされたのでCLOSEします。

closed 2024-04-11 by 200km