ut-issl/s2e-core #467: In the output csv, time is displayed as 60 seconds when the number of seconds is exactly 0.

shunsuke-shimomura, 2023-08-11:
## Overview
In the output csv, time is displayed as 60 seconds when the number of seconds is exactly 0.

### Conditions of occurrence
It always happens when I output CSV file and wherever the number of seconds in simulation comes to exactly 0

###  Additional information
I have not been able to determine it is when the logger outputs it or it is already buggy internally.

## Scope of influence
It depends on where the bug is occurring

## Note
- Set `priority` label if possible.
- Assignees should be set if possible.
- Write a conclusion when closing the issue.

200km, 2023-08-11:
@shunsuke-shimomura バグ報告ありがとうございます！  
これは、logファイルの`time[UTC]`で、`12:01:00.0` と表示されるべきなのに`12:00:60.0`と表示されてしまっている　という意味でしょうか？

shunsuke-shimomura, 2023-08-11:
完全にそれです

200km, 2023-08-11:
`time[UTC]`の元の変数は他の場所ではほとんど使われていないので、ログ表示だけの問題だと思います。  
下記で内部で使っているユリウス日からカレンダー表記に直していますが、大元の変換関数は外部ライブラリから取ってきているので、その仕様で`1分0秒`が`0分60秒`になっているようにも見えますね。

https://github.com/ut-issl/s2e-core/blob/ca493522931f6899ee6f2cd446c5312d74ac119a/src/environment/global/simulation_time.cpp#L223

もう少し確認してみます。

200km, 2023-08-14:
簡単にですが、修正しました。必要なら確認してください。  

https://github.com/ut-issl/s2e-core/pull/468

closed 2023-08-22 by 200km