ut-issl/s2e-core #655: Add connection between electrical power and thermal calculation

200km, 2024-07-18:
## Details
Add the following features.
- Calculate thermal power input depending on the power state of a component.
- Calculate the temperature of a component.

## Closing conditions
Implement the above features.

## Supplementary information
NA

## Note
- Provide a conclusion before closing the issue.

200km, 2024-12-25:
まずは、https://github.com/ut-issl/s2e-core/tree/feature/connect-power-thermal　ブランチにマージする形で進める

kai0722, 2025-04-13:
元になるブランチが消えていたようなので、改めて
https://github.com/ut-issl/s2e-core/tree/feature/connect-power-thermal
を作成。

以後はこちらのブランチに対して部分的な機能の追加等を実施していくことにする。

kai0722, 2025-04-13:
`settings/sample_satellite/thermal_csv_files/node.csv` に、「power port」的な列を増やし、電源状態を紐づけたいコンポ（ノード）はその項目に対応する`power_port`が識別可能な値を設定することで、csvの初期読み込み時にそのノードのheatloadを`power_port`から参照するか、heatload.csvから参照するか（これまで通り、デフォルト）を判断する、という設定が良さそうに思える

kai0722, 2025-04-16:
> 元になるブランチが消えていたようなので、改めて https://github.com/ut-issl/s2e-core/tree/feature/connect-power-thermal を作成。
> 
> 以後はこちらのブランチに対して部分的な機能の追加等を実施していくことにする。

このdevelopから切ったブランチで作業しようかと思ったが、s2e-userでpower_portをオーバーライドするなどして定義して動作確認するほうが後々やりたいことができると気付いたので、s2e-userからの実行と相性の良い最新バージョンv8.0.4からブランチを切って作業をすることにしたい。