ut-issl/s2e-core #664: Add orbit interpolation method as an Orbit calculation feature

200km, 2024-07-24:
## Details
We need a satellite orbit calculation method using interpolation with orbit time series input.  
We already have a basic function as follows.

https://github.com/ut-issl/s2e-core/blob/develop/src/math_physics/orbit/interpolation_orbit.hpp

## Closing conditions
The feature is implemented,

## Supplementary information
NA

## Note
- Provide a conclusion before closing the issue.

fukudakazuya, 2024-09-06:
@200km こちらのタスクの意図としてはTrigonometric以外の手法を追加するということでよろしいでしょうか？

200km, 2024-09-06:
ライブラリは準備しているのでそれを使って次のような機能を実装します。

- 現在様々な軌道伝搬手段で衛星軌道伝搬ができるが、それに「軌道定義ファイルを入力し伝搬する」という機能を付け加える
- 軌道定義ファイルの性格上、必ず内装が必要なので、ライブラリの内装機能を使って実装する
- 軌道定義ファイルとしては標準という意味ではSP3が良いように思うが、もっと単純なCSVファイルでもよく、そこは川端さんなど利用者側と相談する

fukudakazuya, 2024-09-06:
かしこまりました．ありがとうございます．
こちら新しいファイルに実装するのが良いでしょうか？

fukudakazuya, 2024-09-09:
@200km 追加で質問です。軌道定義ファイルで時系列データを読み込んで、それを内挿で補間していくという理解でよろしいでしょうか？伝搬と聞いてRK4などで伝搬するのかなと思ったのですが、今回はそういう意味ではない感じでしょうか。

200km, 2024-09-11:
> こちらどこに実装するのが良いでしょうか？

ソースコードとしてはこの場所が良いと思います。
https://github.com/ut-issl/s2e-core/tree/develop/src/dynamics/orbit

> 軌道定義ファイルで時系列データを読み込んで、それを内挿で補間していく

はい。漢字間違っていましたが内装とかいていたのは内挿のことです。

> RK4などで伝搬するのかなと思ったのですが、今回はそういう意味ではない感じでしょうか。

有力ファイルには推進計操作なども含んだものになるので、RK4などは適しません。なので、新たな軌道伝搬方法（計算方法）を追加するということです。

200km, 2024-09-11:
GNSS衛星の軌道計算をSP3ベースでやっている部分など参考にしてください。

https://github.com/ut-issl/s2e-core/blob/562acf9e4f7a51ffb3b3e0560a423ecc093bbcc0/src/environment/global/gnss_satellites.hpp#L116

fukudakazuya, 2024-09-20:
[image]
[image]

closed 2024-10-01 by 200km